dresden-elektronik/deconz-rest-plugin #2297: Cannot rename sensor

XerXes777, 2020-01-09:
[image]

I can rename the plug by the lights,
but i cant rename the sensor ( power consumption) from my innr sp120 plug

Smanar, 2020-01-09:
I think for the moment the only way is using the API https://dresden-elektronik.github.io/deconz-rest-doc/sensors/ for sensor, for light you can use the old webapp.

XerXes777, 2020-01-09:
okay thanks and the next is...
why i have 2 entrys in iobroker for one innr smart plug in the sensors dir?
by example the picture sensor id 8 and 9 is the same smart plug,
becaus i have actually one smart plug connectet to deconz.

Smanar, 2020-01-09:
It's because consumption and current are not on the same cluster.
Deconz create 1 entry by sensor, if you have the Xiaomi multi sensor, you will have 3 entries, temperature/barometer/pression.

XerXes777, 2020-01-10:
thank you verry much !!
it can be closed.

closed 2020-05-16 by stale[bot]