dresden-elektronik/deconz-rest-plugin #2298: Add support for Tuya TM001-ZA Door/window sensor

sparkydave1981, 2020-01-09:
Please add support for these sensors.  Currently they are not detected during pairing attempts.

https://fccid.io/2ANDL-TM001ZA/User-Manual/User-Manual-4229091

aidbish, 2020-01-09:
Hey @sparkydave1981 
Where can these be purchased as they are small enough to be alternatives to the xiaomi ones

SwoopX, 2020-01-10:
You must have it successfully paired before any device support can be provided.

sparkydave1981, 2020-01-11:
> Hey @sparkydave1981
> Where can these be purchased as they are small enough to be alternatives to the xiaomi ones

I bought them from [here](https://greenhse.com/).  They are tiny!  It would be awesome if they could work with deCONZ

sparkydave1981, 2020-01-11:
> You must have it successfully paired before any device support can be provided.

So since I can't get it to pair I guess there is nothing I can do... I have two of them so I will try the second one just to be sure.

SwoopX, 2020-01-11:
How did you try pairing, from deconz GUI or Phoscon? Make sure to give both a try.

sparkydave1981, 2020-01-12:
> How did you try pairing, from deconz GUI or Phoscon? Make sure to give both a try.

I only used deCONZ within Home Assistant.  How do I use Phoscon?  That said, I am running HassIO on top of Ubuntu so I wonder if there is something extra I need to do to ensure HassIO has the correct access to the Conbee 2...  I will try to install deCONZ on the underlying Ubuntu and see how I go there.

closed 2020-05-18 by stale[bot]

sparkydave1981, 2020-07-01:
I ended up using these devices with the inbuilt ZHA component in Home Assistant rather than using DeConz and they work fine like that.  DeConz was simply not working with them

dpozimski, 2021-03-23:
I am refreshing this issue.
Is it possible to add  support for this device? 
Problem still exists. Deconz paired with tuya door sensor correctly but the sensor is not visible in phoscon completely.

Mimiix, 2021-03-23:
@dpozimski 
Is it working with the API? 
Yes: Open a request in the Phoscon repo > https://github.com/dresden-elektronik/phoscon-app-beta
No: Open a device request issue.  > https://github.com/dresden-elektronik/deconz-rest-plugin/issues/new?assignees=&labels=Device+Request&template=device-request.md&title=Device+name