dresden-elektronik/deconz-rest-plugin #2300: Add support for Xiaomi Mijia Smart Light Sensor (GZCGQ01LM)

jeroenvdheuvel, 2020-01-09:
Unfortunately, I'm unable to find a link on an official website. This is the actual device: 

[image]

Here is a link to seller: https://aliexpress.com/item/4000475208359.html

Information about the device:
[image]

[image]

[image]

jeroenvdheuvel, 2020-01-09:
What's the best way for me to start implementing this sensor? Is there documentation that would help me get up to scratch?

Smanar, 2020-01-10:
Depend of your luck ^^.

- Be sure it's working on deconz, correct pairing, can read information, can send action to it, ect ...
- Add it in "white list" if needed. This part is mostly for sensor
- Recompile the code, replace the file and restart deconz, the process is described on github mainpage https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
- Submit a Pull.

If you are lucky it will be enought else describe your problem, perhaps someone will have idea.

You can find usefull informations and one exemple on this issue https://github.com/dresden-elektronik/deconz-rest-plugin/pull/2002

Alexxander0, 2020-02-05:
Hello,
did you find any way to pair it and make it work?

xCruziX, 2020-02-11:
A status update would be great.

tjaena, 2020-02-11:
There is a duplicate issue (#2380) to this one. Which resulted in a pull request (#2375), which was already merged to master.

As of my understanding, you could either try to compile from the source or wait for a new release with the support for the sensors.

I’ll try the first one at a later time this week and see if it works out.

jeroenvdheuvel, 2020-03-10:
Like @Alexxander0 I'm having trouble pairing the sensor.

It's being recognized now:
[image]

The console output:
[image]

But when I click the "ready" button, the sensor is not there.

sjee105, 2020-03-10:
It doesn't showup in phoscon but you should see it on your network map. I have two working with HA.

jeroenvdheuvel, 2020-03-10:
@sjee105 you're right I also see the sensor in HA and it works fine.

I'm not sure if this issue should still be kept open. The pairing is confusing but it does work in the end.

SwoopX, 2020-03-10:
If I understand the release notes of .75 correctly, the sensor should now also show up in Phoscon. Afaik, it is fully supported now.

@jeroenvdheuvel What's exactly confusing from the pairing perspective?

jeroenvdheuvel, 2020-03-11:
The pairing works fine. But after pairing the sensor isn't visible in the Phoscon web interface. That means that there's no way to change the name.

You can see this in the recording I made:
[image]

SwoopX, 2020-03-12:
Ok. I see. One needs to understand that Phoscon is decoupled from deconz in terms of device support. Phoscon is closed source and therefore, new devices must not automatically be shown. Sad, but true.

However, the green success message is what's important.

Janky0, 2020-04-28:
But how can i use this sensor in FHEM when its not shown in deconz?

SwoopX, 2020-04-28:
It depends a bit if in FHEM stuff is created automatically or not. If the sensor is paired correctly, it will be visable via REST API and therefore also visible for FHEM. You may want to check your deconz device there and get the list of sensors (don't think anything's available as light for the device). Then you require the corresponding ID to create a new device inside FHEM (or multiple in case more than one sensor is created for the Xiaomi by deconz).

Janky0, 2020-04-28:
yes thats no problem but i get no lux or something like that

`Internals:
   CFGFN     
   DEF        sensor 2  IODev=deCONZ
   FUUID      5ea85d71-f33f-0688-81a1-3b7723ced27c58e5
   FVERSION   31_HUEDevice.pm:0.217030/2020-04-16
   ID         S2
   INTERVAL   
   IODev      deCONZ
   NAME       Lichtsensor
   NR         738
   STATE      Initialized
   TYPE       HUEDevice
   manufacturername LUMI
   modelid    lumi.sen_ill.mgl01
   name       Lichtsensor
   on         1
   reachable  1
   swversion  20191118
   tholddark  12000
   type       ZHALightLevel
   uniqueid   04:cf:8c:df:3c:77:26:ae-01-0400
   helper:
     devtype    S
     reachable  0
     update_timeout 1
     configList:
     json:
       ep         1
       etag       14a2900bb5e3ab4c90316cf9defb8124
       lastseen   2020-04-28T16:44:47.781
       manufacturername LUMI
       modelid    lumi.sen_ill.mgl01
       name       Lichtsensor
       swversion  20191118
       type       ZHALightLevel
       uniqueid   04:cf:8c:df:3c:77:26:ae-01-0400
       config:
         battery   
         temperature 0
         tholddark  12000
         tholdoffset 7000
       state:
         dark       
         daylight   
         lastupdated none
         lightlevel
         lux       
     setList:
Attributes:
   IODev      deCONZ
   model      lumi.sen_ill.mgl01`

ragefan, 2020-05-20:
Same issue with same captor. I compiled from source. Anything i can do to collect logs ?
The captor seems detected but no value from the sensor.

ragefan, 2020-05-20:
> 
> 
> @sjee105 you're right I also see the sensor in HA and it works fine.
> 
> I'm not sure if this issue should still be kept open. The pairing is confusing but it does work in the end.

Hi,
Can you read the sensor value ?
I cannot in Jeedom.

SwoopX, 2020-05-20:
@ragefan we could do pretty much the same exercise here as for the smart plug. Might be that here's still some room for improvement as I don't know (yet) what's been reported.

ragefan, 2020-05-20:
Let's go : can you please have a look to those logs ? The sensor is inside.
[putty.log](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4658884/putty.log)

Screenshot from deconz have already been posted at the beginning of the thread.

Smanar, 2020-05-20:
Can try to make manual binding ? https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2380#issuecomment-588213553

I know it's already in the code, but we can see in log
>18:47:33:783 MAC poll fastEnddeviceProbe() 0x04CF8CDF3C78815D
18:47:33:783 don't create binding for attribute reporting of sensor Light Sensor

ragefan, 2020-05-20:
Is it saved somewhere ? I was using a headless device up to now.

SwoopX, 2020-05-20:
@ragefan You require deconz GUI to set up binding and reporting manually. However, we could also do that via a code change for testing purposes. In bindings.cpp, search for that line
`sensor->modelId().startsWith(QLatin1String("lumi.plug.maeu01")) ||`
and change it to
`sensor->modelId().startsWith(QLatin1String("lumi.")) ||`

I've introduced general binding for Xiaomi with version .74, but reverted as people started to report strange issues with the motion sensor. As I don't have the devices in hand, I simply do not know if they support binding and attribute reporting or not. However, as this is an Aqara ZB3.0, it seems it does, so we probalby should include it as well.

Smanar, 2020-05-20:
I have see a screenshoot from you ? Was just for some test ?
And good question ^^. Some information are stored on the device (reporting values), but I have a doubt for "binding table".

@SwoopX have perhaps an answer ?
All xiaomi device have binding/report disabled because Xiaomi use it's own notifications, perhaps it's somethinf different for this device ?

ragefan, 2020-05-20:
Yes, just for some tests under windows (not "production" env)

ragefan, 2020-05-20:
[image]

Just a screenshot from HA, showing "unknown" value for light sensor

Smanar, 2020-05-20:
I m searching information ATM, but this device is zigbee 3.0, so perhaps a better one with other method than classic Xiaomi.

SwoopX, 2020-05-20:
One thing to note: Aqara != Mija

The Aqara ZB3.0 devices seem to be closer to "normal and expected" behaviour than the ZB3.0 Mija's.

ragefan, 2020-05-20:
Hum ... is this helping ?
https://github.com/Koenkk/zigbee2mqtt.io/blob/master/docs/devices/GZCGQ01LM.md

Smanar, 2020-05-20:
OK so it s that
This device need binding.

We need to bind the cluster 0400

ragefan, 2020-05-20:
Thanks ! Sorry but i don't know what "bind the cluster" mean :-(
Can it be done using some command lines ?
Where is it store ? if in the device, is there a risk to crash it ?

SwoopX, 2020-05-20:
@Smanar That's not adequate, I'm afraid. Take the sensors (temp, door, etc.). The round one (Mija) are crappy and unstable as hell. The Edged ones (Aqara) are much more reliable, although far from being perfect.

Now, in ZB3.0 land, the Mija smart plugs still use that analog cluster stuff, have their embedded attribute reporting, manufacturer specific cluster and the special Xiaomi attribute reporting thingy. With the ZB3,0 sensor here and also with the smart plug, you seem to have what everybody else expects and uses: binding and attribute reporting, NO Xiaomi special attributes as it seems.

SwoopX, 2020-05-20:
@ragefan Recomplie with the proposed change. It's way easier with your headless setup.

ragefan, 2020-05-20:
which one ? is there a pull request somewhere ?

SwoopX, 2020-05-20:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2300#issuecomment-631716756

ragefan, 2020-05-20:
Ouch, just missed that one ! Thanks, will try that :)

SwoopX, 2020-05-20:
Np

ragefan, 2020-05-20:
Well well well, that's sound better as i get new attributes :
[image]

However, it seems i still cannot collect the intensity of light ?
How often is it refreshed ?

SwoopX, 2020-05-20:
Please allow me. You now patched what I reverted some days ago :)

ragefan, 2020-05-20:
:) come on! This was a nice working patch. :) Do you need more clues in order to put it back again?

Indeed that one is working now 😁 Thanks!

SwoopX, 2020-05-20:
😄 Just wasn't explicit enough. All done.

ragefan, 2020-05-21:
Thanks! I guess we can close that one as well then 😉 Go for the next one, I am waiting some brand new xiaomi sensors.

DukeNuken, 2020-05-26:
Guys. I have the same sensor from aliexpress and it does not show up in deconz. So what should be done here? Will it be added to deconz addon after some update?

ragefan, 2020-05-27:
Hi! You need to build from source or wait the next release. It is fixed in the current source.

SwoopX, 2020-05-27:
What is your deconz version? Do you really mean it doesn't show up in deconz or rather Phoscon?

DukeNuken, 2020-05-27:
I am using HassOS 3.13 with HomeAsistastant and there are installed deconz addon. 
[image]
When I am adding the new sensor with the Phoscon web interface it shows sensor added but there are no new devices in the list.

SwoopX, 2020-05-27:
Thanks, so the device should be fully functional. This repo is about the REST API, not Phoscon.

However, for Phoscon, they are putting quite some effort in catching up right now.

ragefan, 2020-05-27:
> I am using HassOS 3.13 with HomeAsistastant and there are installed deconz addon.
> [image]
> When I am adding the new sensor with the Phoscon web interface it shows sensor added but there are no new devices in the list.

I ve got same issue with phoscon but as said this is another topic.

Mimiix, 2020-06-06:
As mentioned before in the other topic / issue: not everything is shown in Phoscon due to the fact it needs to be added manually.

closed 2020-06-06 by Mimiix

bcutter, 2021-02-17:
So is this device fully supported (deCONZ/Phoscon)?

Meaning sensor values are provided (battery state, illuminate).

SAOPP, 2021-02-24:
> So is this device fully supported (deCONZ/Phoscon)?
> 
> Meaning sensor values are provided (battery state, illuminate).

Yes, I have been using the device for almost a year.

[image]

goegsig, 2021-03-22:
> 
> 
> > So is this device fully supported (deCONZ/Phoscon)?
> > Meaning sensor values are provided (battery state, illuminate).
> 
> Yes, I have been using the device for almost a year.
> 
> [image]

How did you pair it ? Mine just shows up like this. Nothing in phoscon.
[image]

SAOPP, 2021-03-23:
U'd better tell me what version of Deconz u have.

goegsig, 2021-03-23:
[image]

SAOPP, 2021-03-23:
Strange, but the sensor should be in Phoscon, u probably have a different revision, but these are just thoughts aloud, probably Phoscon doesn't know your revision... try to reconnect it.