dresden-elektronik/deconz-rest-plugin #2301: Add support for KlikAanKlikUit Smoke Sensor

mathim, 2020-01-09:
Hello

I would like to request support for device support for
- KlikAanKlikUit Smoke Sensor
- ID ZSDR-850

Device looks pretty much the same as the Heiman HS1SA

[image]
[image]
Thanks in advance
Kr

SwoopX, 2020-01-10:
According to the code and info provided, it should already be supported.

ebaauw, 2020-04-05:
Can the siren on this sensor be activated over ZigBee.  Does the device have a working _IAS WD_ cluster?

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix