dresden-elektronik/deconz-rest-plugin #2302: Aqara Smart Wall Plug: Power measurement

ulug79, 2020-01-10:
lumi.ctrl_86plug.aq1 work in phoscon, but power sensors absent.

SwoopX, 2020-01-13:
Does it support power measurement? What are the clusters read?

ulug79, 2020-01-14:
[image]

As I understand it, the outlet itself does not give data on energy consumption? in mihome power consumption shows.

ulug79, 2020-01-15:
[image]

It turns out that this outlet does not have a separate cluster for measuring consumption. But there is a separate cluster for this

ebaauw, 2020-01-15:
The Aqara wired switches use _Analogue Input (Basic)_ on the other endpoints for reporting consumption and power.  Good chance that this plug does as well.  Consumption and power are probably also included in the Xiaomi special attribute report.

SwoopX, 2020-01-15:
To me, it looks very related to #2261, just with another model name...

ulug79, 2020-01-26:
> The Aqara wired switches use _Analogue Input (Basic)_ on the other endpoints for reporting consumption and power. Good chance that this plug does as well. Consumption and power are probably also included in the Xiaomi special attribute report.

5:04:06:032 0x00158D000239618E extract Xiaomi special attribute 0xFF01
15:04:06:032    64 on/off 1
15:04:06:032    03 temperature 24 °C
15:04:06:032    98 power (?) 0x00000000
15:04:06:032    95 consumption (?) 0x36A588F7
15:04:06:032    05 RSSI dB (?) 1 (0x0001)
15:04:06:032    9a unknown 0 (0x00)
15:04:06:032    08 unknown 4383 (0x111F)
15:04:06:032    07 unknown 0 (0x0000000000000000)
15:04:06:032    09 unknown 512 (0x0200)

ulug79, 2020-01-26:
Plug very often report about attribute of cluster

15:01:47:023 ZCL attribute report 0x00158D000239618E for cluster 0x000C, ep 0x02
15:02:02:030 ZCL attribute report 0x00158D000239618E for cluster 0x000C, ep 0x02

SwoopX, 2020-01-26:
That is power or consumption reporting.

ulug79, 2020-01-26:
> That is power or consumption reporting.

In this plug two parameters reported (in Mihome), power and consumption.

mcallan83, 2020-02-12:
I am seeing the same problem. I'd love to dig in and come up with a fix, but I am brand new to deconz and wouldn't know where to start. I can verify, that I can see the correct wattage being reported in the Analogue Input (Basic) section. (1158.07 watts)

[image]

SwoopX, 2020-02-12:
@ebaauw I don't see any "lumi.ctrl_" as supported device. Wouldn't that make sense regarding the absense of the sensors?

ebaauw, 2020-02-12:
Where are you looking?  What are you looking for?

```
$ grep -n ctrl_ *
database.cpp:3309:            if (!sensor.modelId().startsWith(QLatin1String("lumi.ctrl_")) &&
de_web_plugin.cpp:170:    { VENDOR_JENNIC, "lumi.ctrl_neutral", jennicMacPrefix }, // Xiaomi Wall Switch (end-device)
de_web_plugin.cpp:176:    { VENDOR_115F, "lumi.ctrl_ln", jennicMacPrefix}, // Xiaomi Wall Switch (router)
de_web_plugin.cpp:1496:        node->nodeDescriptor().manufacturerCode() == VENDOR_JENNIC || // Xiaomi lumi.ctrl_neutral1, lumi.ctrl_neutral2
de_web_plugin.cpp:1673:                                // lumi.ctrl_neutral1 and lumi.ctrl_neutral2 have more 5 endpoints
de_web_plugin.cpp:1678:                                // blacklist switch endpoints for lumi.ctrl_neutral1 and lumi.ctrl_neutral2
de_web_plugin.cpp:1679:                                DBG_Printf(DBG_INFO, "Skip load endpoint 0x%02X for 0x%016llX (expect: lumi.ctrl_neutral1 / lumi.ctrl_neutral2)\n", i->endpoint(), node->address().ext());
de_web_plugin.cpp:1715:                            // Xiaomi lumi.plug and wall switch lumi.ctrl_ln1.aq2, lumi.ctrl_ln2.aq2
de_web_plugin.cpp:1721:                            // Xiaomi wall switch lumi.ctrl_neutral1, lumi.ctrl_neutral2
de_web_plugin.cpp:1722:                            // TODO exclude endpoint 0x03 for lumi.ctrl_neutral1
de_web_plugin.cpp:3836:                    else if (modelId.startsWith(QLatin1String("lumi.ctrl_neutral")))
de_web_plugin.cpp:4025:                    else if (modelId.startsWith(QLatin1String("lumi.plug")) || modelId.startsWith(QLatin1String("lumi.ctrl_ln1")))
de_web_plugin.cpp:4036:                    else if (modelId.startsWith(QLatin1String("lumi.ctrl_ln2")))
de_web_plugin.cpp:4060:                    else if (modelId.startsWith(QLatin1String("lumi.ctrl_ln")) && i->endpoint() == 0x05)
de_web_plugin.cpp:5045:        if (!sensorNode.modelId().startsWith(QLatin1String("lumi.ctrl_")) &&
de_web_plugin.cpp:5648:                        i->modelId().startsWith(QLatin1String("lumi.ctrl_neutral")) ||
de_web_plugin.cpp:5649:                        (i->modelId().startsWith(QLatin1String("lumi.ctrl_ln")) && event.clusterId() == MULTISTATE_INPUT_CLUSTER_ID) ||
de_web_plugin.cpp:6273:                                    else if (i->modelId().startsWith(QLatin1String("lumi.ctrl_neutral")))
de_web_plugin.cpp:6650:                                         i->modelId().startsWith(QLatin1String("lumi.ctrl_")))
de_web_plugin.cpp:6752:                                else if (i->modelId().startsWith(QLatin1String("lumi.ctrl_ln")))
de_web_plugin.cpp:8614:    if (!found) // not all light endpoints have a group cluster (lumi.ctrl_ln2.aq1)
de_web_plugin.cpp:9702:        else if (tag == 0x07 && dataType == deCONZ::Zcl64BitUint) // lumi.ctrl_ln2
de_web_plugin.cpp:9706:        else if (tag == 0x08 && dataType == deCONZ::Zcl16BitUint) // lumi.ctrl_ln2
de_web_plugin.cpp:9710:        else if (tag == 0x09 && dataType == deCONZ::Zcl16BitUint) // lumi.ctrl_ln2
de_web_plugin.cpp:9723:        else if ((tag == 0x64 || structIndex == 0x01) && dataType == deCONZ::ZclBoolean) // lumi.ctrl_ln2 endpoint 01
de_web_plugin.cpp:9748:        else if (tag == 0x65 && dataType == deCONZ::ZclBoolean) // lumi.ctrl_ln2 endpoint 02
de_web_plugin.cpp:9763:        else if (tag == 0x6e && dataType == deCONZ::Zcl8BitUint) // lumi.ctrl_neutral2
de_web_plugin.cpp:9767:        else if (tag == 0x6f && dataType == deCONZ::Zcl8BitUint) // lumi.ctrl_neutral2
de_web_plugin.cpp:9771:        else if (tag == 0x95 && dataType == deCONZ::ZclSingleFloat) // lumi.ctrl_ln2
de_web_plugin.cpp:9783:        else if (tag == 0x98 && dataType == deCONZ::ZclSingleFloat) // lumi.ctrl_ln2
de_web_plugin.cpp:9791:        else if (tag == 0x99 && dataType == deCONZ::Zcl32BitUint) // lumi.ctrl_neutral2
de_web_plugin.cpp:9795:        else if (tag == 0x9a && dataType == deCONZ::Zcl8BitUint) // lumi.ctrl_ln2
de_web_plugin.cpp:9807:        else if (tag == 0x9b && dataType == deCONZ::Zcl16BitUint) // lumi.ctrl_neutral2
de_web_plugin.cpp:9839:        if (lightNode.modelId().startsWith(QLatin1String("lumi.ctrl_neutral")))
de_web_plugin.cpp:9855:        else if (lightNode.modelId().startsWith(QLatin1String("lumi.ctrl_ln")))
de_web_plugin.cpp:9944:            // DBG_Assert(item != 0); // expected - no, lumi.ctrl_neutral2
de_web_plugin.cpp:16391:             sensor->modelId().startsWith(QLatin1String("S2")) ||sensor->modelId().startsWith(QLatin1String("lumi.ctrl_")))
```
Note: this is my local fork; the line numbers might be off.

SwoopX, 2020-02-12:
de_web_plugin.cpp very top, line 170 or so. But haven't checked the rest since I was in a rush.

SwoopX, 2020-03-05:
@mcallan83 from which endpoint was the screenshot taken, 02 or 03? Consumption should be 03.

mcallan83, 2020-03-07:
@SwoopX my screenshot was from 02.

03 looks very similar, but for the Present value, it seems to have a static value of .54xxxx even if the plug is off or on.

SwoopX, 2020-03-07:
@mcallan83 Thanks for coming back in this. The static value seems to be an indication that consumption is not supported. Not sure if you read the attributes but if you could note the value after reading and keep an eye on it for a couple of days in how far it changes (if at all)...

mcallan83, 2020-03-07:
Ok... will do. For 02, though, it is the correct wattage that is being shown. Would there be any way to get that working?

SwoopX, 2020-03-07:
Just not to mix up. 02 gives the current power usage, 03 is supposed to give you the power consumed (so consumption) over the time since the last device reset.

My understanding was that power was working all the time...

mcallan83, 2020-03-07:
Yes you are correct. 02, shows wattage and updated is real time. 03 is stuck at a static value, but I will monitor it for changes.

jomar, 2020-03-10:
I experience the same missing power consumption for my Xiaomi plugs.  Current power usage is fine.  Opening the cluster info for 03 showed 0 for "Present value".  Manually tapping "read" above the attributes seems to update with the correct value.

SwoopX, 2020-03-10:
@jomar do you have exactly the same modelID?

jomar, 2020-03-10:
No, my devices are detected with model identifier: "lumi.plug".

SwoopX, 2020-03-10:
@jomar you may want to try out a manual binding like so https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2373#issuecomment-595414912

Take the analog input cluster from endpoint 03 as source. Afterwards, you may want to set attribute reporting by double clicking on the present value, set values 1, 60, 1 and write the configuration. Eventually, that does the trick.

jomar, 2020-03-11:
I was able to successfully bind the endpoint, but when trying to write config for reporting, it gives the message: UNREPORTABLE_ATTRIBUTE.
[image]

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix

maskiopt, 2021-03-03:
Hi all,
I also have this plug and I can only control ON/OFF, there's no power consumption info.
Please add support for this device.

Thanks

alextud, 2021-03-04:
Having a consumer for the plug might do the trick

see https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2583#issuecomment-764951311

maskiopt, 2021-03-04:
@alextud I've tried that.
Didn't do anything for me. Still only "on/off"

alextud, 2021-03-04:
You can see if plug sensors are visible in deconz using REST API https://dresden-elektronik.github.io/deconz-rest-doc/endpoints/sensors/

For me they did appear here after I re-added a couple of times.

Also changing "Reporting interval" from 300 seconds (5 minutes) to x seconds will update sensor more often ....

[image]