dresden-elektronik/deconz-rest-plugin #2303: Create, update a virtual sensor with the rest API to use in deCONZ

feikede, 2020-01-10:
I'd like to create a "virtual" sensor for my photovoltaic production. My pv has a request api so I am all set to create a sensor in deCONZ. I do

```
POST phoscon/api/5xx7D914/sensors
{
    "config": {
        "on": true
        "reachable": true
    }
    "manufacturername": "Feik"
    "modelid": "T1000"
    "name": "PowerUse"
    "swversion": "1.0"
    "type": "CLIPGenericStatus"
    "uniqueid": "0x001faa0000000001"
}
[
    {
        "success": {
            "id": "2"
        }
    }
]
```
and

```
PUT phoscon/api/5xx7D914/sensors/2/state
{
  "status": 2342
}
[
    {
        "success": {
            "/sensors/2/state/status": 2342
        }
    }
]
```
To check that
```
GET phoscon/api/5xx7D914/sensors
{
    "1": {
        "config": {
            "configured": true,
            "on": true,
            "sunriseoffset": 30,
            "sunsetoffset": -30
        },
        "etag": "a3e0b40dcef13cb4bcec581c532b73d7",
        "manufacturername": "Philips",
        "modelid": "PHDL00",
        "name": "Daylight",
        "state": {
            "dark": false,
            "daylight": true,
            "lastupdated": "2020-01-10T11:19:59",
            "status": 170,
            "sunrise": "2020-01-10T07:00:14",
            "sunset": "2020-01-10T15:39:27"
        },
        "swversion": "1.0",
        "type": "Daylight",
        "uniqueid": "00:21:2e:ff:ff:03:cd:5c-01"
    },
    "2": {
        "config": {
            "on": true,
            "reachable": true
        },
        "etag": "2253fe5f0ccd5b2b74850612d770b497",
        "manufacturername": "Feik",
        "modelid": "T1000",
        "name": "PowerUse",
        "state": {
            "lastupdated": "2020-01-10T13:23:22",
            "status": 2342
        },
        "swversion": "1.0",
        "type": "CLIPGenericStatus",
        "uniqueid": "0x001faa0000000001"
    },
}
```

How can I make my virtual "PowerUse"-Sensor usable in deCONZ or PhosconAPP?
Finally, I'd like to create a rule to start a power plug when I'am producing enough solar power.

Thanks, Rainer

ebaauw, 2020-01-10:
The CLIP sensors live in the REST API plugin; they won’t ever be visible in the deCONZ GUI.  Your API client, i.c. Phoscon would need to support CLIP sensors; I doubt that Phoscon does so, but homebridge-hue exposes all of them, except CLIPSwitch (since there is no way to specify what button events it could generate).  For power use, you’d better create a CLIPConsumption and/or CLIPPower sensor, so API clients can understand what it’s reporting (en lieu of a generic status).  Anything you do in rules with ZigBee sensors also works for CLIP sensors.  Even more, since you can change the value of a CLIP sensor from rules.

A while ago, I did some write-up on CLIP sensors, but I haven’t maintained it.  See https://github.com/ebaauw/homebridge-hue/wiki/Getting-Started#24-clip-sensors.

feikede, 2020-01-11:
Thanks ebaauw. Unfortunately I don't have any apple devices here. Is there any other way to create a "virtual" sensor for my photovoltaic production that can be used by deConz or Phoscon? It seems to may like a possible task because there's a "virtual" daylight already in phoscon and deCONZ (see #1 above).

ebaauw, 2020-01-11:
I repeat: you cannot create virtual devices in deCONZ; the Daylight sensor lives in the REST API plugin.  It’s hard-coded there.  In theory, you could create a PR implementing another virtual sensor like that, but I think CLIP sensors are powerful enough for what you eant.  You can create CLIP sensors using the REST API, no Apple devices needed for that.  To display CLIP sensors, you need an API client that plays nice with them.  I don’t think the Phoscon web app does.  It’s not open source, so only dresden elektronik can change this.

closed 2020-05-17 by stale[bot]