dresden-elektronik/deconz-rest-plugin #2304: Ikea Color Bulbs don't react to Alexa color changes - the second

acid115, 2020-01-10:
Hi, currently i have an issue, that my Ikea Color Bulb ignores color changes, triggered by an Amazon Echo 3.gen (through speech or app)
Error is always: "malfunction of the device". Switch on/off and Change brightness works.

The author of the issue #1866 (https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1866) solved it by installing 2.05.69 and repairing the bulbs.

i am currently on 2.05.71 / 26330500 with my ConBee I
Bulb: 
IKEA of Sweden
Modell TRÅDFRI Lampe E27 CWS opal 600lm
Version 1.3.009

Yesterday i delete the bulb (in Phoscon and Alexa App), reset the bulb and reconnect the bulb.
the bulb gets a new id, but the error didnt disappear. 

i have debugged the little bit and can see, that the amazon echo sends this command to the rest api:
"on": true
"hue": 18699 (example)
"sat": 254

but if a send this command though an browser oder curl request, the bulb do not change the color. The color is only be changed, when i send x,y values:
example:
"on": true
"bri": 200
"x,y": "[0.2322,0.6422]"

Pirlobeck, 2020-01-10:
Hallo,
I can confirm the same behavior when using an Alexa Echo Dot Gen2. When I  change the color of my Osram an Philips bulbs, the color change works, but I get the answer from Alexa "it looks like, that the bulb does not work proper!"

acid115, 2020-01-12:
I have found the problem and now i am looking for a solution.
I am now able to compile the plugin and can change the source.
I have tried to calculate the xy values from the given hue and sat.
There are some lines the code where the xy is calculated (Hsv2Rgb,Rgb2Xy) but the result is wrong. These are not the xy values for the given colors / hue values.
I set the wrong calculated xy-values for the light, that is ok. But the color is wrong.
So i am trying to find out the right calculation for hue - xy transformation ...

ebaauw, 2020-01-12:
The `hue` and `sat` values are utterly useless, since they represent different colours to different types of lights (even of the same vendor, see https://developers.meethue.com/develop/get-started-2/core-concepts/).  Only `xy` and `ct` values are standard across light types, but even then, not every light type can physically reproduce every colour, so they same `xy` value might look differently on different types of lights.  Colour is a bitch.

The `xy` values express colour in the CIEE 1931 colour space, which is different from the RGB colour space that your PC or phone uses.  The lines of code translate colours in the RGB colour space (which might be expressed as HSV or RGB) to equivalent CIEE 1931 colours, so the lights more or less show the same colour as you selected from your PC or phone app.

acid115, 2020-01-13:
thank you for your explanation.
but i think there is a bug in Hsv2Rgb and Rgb2Xy.
you need three values: the real Hue (0-254), the Saturation (always 1.0 because the echo sends only 254) and v (also constant 1.0). Then i take some different Hue values (169 Blue, 42 yellow, 84 green) and after Hsv2Rgb and Rgb2Xy the values for x and y differ not so much.
x is always between 0.6900 and 0,6999 and y between 0.3300 and 0.3399 (i dont know exactly)
So there is no  significantly color-change. there must be  bug.

so i have think about it and find a quick solution for me, we can check the calculation later.
the Echos know only a set of colours (maybe 20)
So there is only a set of hue-values, the echos sent.
for each colour i have to find out the correct x-y value by setting the colour with the slide of the phoscon app and get the x-y value through the web api (get request of the current state).

i let calculate the real hue value (value between 0-255 instead 0-65535) and say:
if hue value = 169 (blue) then set x and y for blue (with the colour slider for this ikea bulb)
and it works :)
But the echo has still the error message: Malfunction of the device  - that is not so nice.
so i found a solution for this too.
After the Echo sets the color and get the success messages, the echo checks the state of the device again and checks hue and sat value. But these values are at the moment not set in the lightnode state - only x and y (because we set only x and y to change the color). So if these values (from the GET state request) differ to the values requested (PUT), the echos throws this error message.
so i set the given hue and sat value for the light node too, after setting x and y.
Now the echo can see the requested hue and sat value, if the echo makes the control-GET-Request of the light state.
Now, everything works (for me and my bulb :) ) - no error message anymore.
But at the moment i have no solution how we can implement this for everyone. 
Or maybe no one has the problem only me...
Maybe i am the only one who uses the Ikea color bulb, the phoscon gateway, and amazon echos with 2nd gen or higher...
The last 2 years i had a Echo 1.gen in my echo network. This Echo holds the api-key and made all of the Push and Get Requests to the phoscon gateway. I dont know exactly but i think, that this echo has sent x y values instead of hue values...
then i switched it off  and now i use only echos 2.gen and higher and colorchange was not possible anymore with echo.

if we can repair the calculation for xy for an given hue / sat, then i think, we can implement the changes for all, bit at the moment it is a little bit dirty.
maybe the calculated x y values are the same like for philips oder osram color bulbs..

acid115, 2020-01-13:
ahhhh.
i have found the error....
theoretically, i will test it this evening...

in rest_lights.cpp you can see in line 1198:
`            f /= 360.0;

            if (f > 1.0)
            {
                f = 1.0;
            }

            hue = f * 254.0;

            DBG_Printf(DBG_INFO, "hue: %u, sat: %u\n", hue, sat);

            double r, g, b;
            double x, y;
            double h = ((360.0 / 65535.0) * hue);
            double s = sat / 254.0;
            double v = 1.0;

            Hsv2Rgb(&r, &g, &b, h, s, v);
`
at line double h = ((360.0 / 65535.0) * hue); 
here you can not take the hue value that is between 0 and 254, here you need the value given from request - the value between 0 and 65535

because so the h-value is between 0 and  1.4 and not between 0 and 360

so, i will test this this evening.

acid115, 2020-01-13:
So, i can confirm now, that everything works fine now.
I have used the given hue value between 0 and 65535 divided by 360 for the hsv2rgb function.
Every color seems correct now, except the yellow value, i had to correct manually in the source to get a real yellow
And, important, set the hue and sat value for the lightnode, because the checks these values after setting.
The same problem i had for my CT bulb 950lm from ikea.
I have to correct the ct value from the echo by adding "65" to reach the min and do not overrun the max value.
And than, important, set the requested ct value for the lightnode again, because echo checks this.
So now i dont have any malfunction alerts and now everything works with the ikea bulbs.
I can push my changes or show my changes and you can build it in where you want.

closed 2020-05-20 by stale[bot]