dresden-elektronik/deconz-rest-plugin #2306: Ikea Tradfri Remote Group Binding Button Actions

marcelmu, 2020-01-10:
In my current I have a Light Group all my lights are assigned to.
In the group's settings I connected my Tradfri 5 Button Remote und assigned different actions to its buttons.
This works fine as is!

The only downside using this solution is, that I'm unable to control my lights when deconz is not running. I'm aware that for this to work, the lights have to be assigned to the group created by the remote.
But in this group I'm not able to assign different actions to the buttons. At least not using the webapp. Is there any way to achive this?

Smanar, 2020-01-10:
Nope, sorry.
You have only 2 possiblities:
- Direct working mode, but using defaut zigbee command.
- Full control but managed by coordinator/application.

marcelmu, 2020-01-10:
Thank you!
To bad.

closed 2020-01-10 by marcelmu