dresden-elektronik/deconz-rest-plugin #2308: Ok, I think I got it finally!

cobch, 2020-01-10:
Ok, I think I got it finally! 
For future newbies (like myself) who arrive in this github like I did:
1) Setup VNC in deconz config and open the GUI
2) You see your zigbee mesh. Open Cluster info tab (bottom left)
3) Click on the second dot next to each light's name and choose Legrand-Specific clusters option
4) Edit (double click) the attribute on the cluster info tab ("0x0101" for enabling Dimmer on and "true" for enabling the led on the switch) and click **write** for both.

It should be working now! Thanks guys, if it weren't for this github, I would have called an electrician (because I thought I did the cabling wrong :P :P )

_Originally posted by @nassost79 in https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-564976964_

cobch, 2020-01-10:
Hi,

I am a newbie and thanks to you I have managed to activate the dimmer function of my Legrand Céliane Switch with dimmer without neutral in home assistant with ConBee. 
My only problem is to activate the dimmer function when I push the switch manually (-:
I only can use on/off when I push on the switch. Any ideas what I should do to make it work?

Thanks in advance for your help.

Smanar, 2020-01-10:
Hello, just for information the second option is not necessary for dimmer, it's just to have the LED always on.

nassost79, 2020-01-10:
@cobch hi! Happy I could help! Unfortunately, I haven't managed to do this (and have not bothered to find a solution for it to be frank).

Smanar, 2020-01-10:
@cobch have you the Legrand gateway ? to update the switch ?

cobch, 2020-01-11:
Hi,
I have updated the switch with Legrand gateway and managed to fix the config above. I can use the dimming function through deconz app or home asssistant. But when I long press the switch manually no dimming is working. I guess I have to do another config either in home assistant or maybe in deconz GUI. Any advices will be welcome with pleasure.

Smanar, 2020-01-11:
Ha ?
Lol, first time I see a device working in deconz and not in the reality.
It have worked on Legrand gateway ?

The dimmer feature is working offline, without deconz or home assistant, so it's realy strange.

You can see some return on websocket when you long press the switch ?

Smanar, 2020-01-11:
An easy thing to try.
Try the dimmer feature but with deconz turned off, to be sure it's not him or HA that trun the switch itself.

cobch, 2020-01-11:
When I used the switch with Legrand gateway I had to choose the option "light dimmer option" in order to make the dimming functions worked when I was pushing on the switch manually. The standard is "dimming" not activated. See picture below
[image]

I then removed the switch from Legrand gateway in order to import it in Deconz. As described above I managed to imported it and make the dimming function work in Deconz/Home assistant, but nothing happen when I long press the switch manually )-:

I tried to stop Hassio (and home assistant) but the dimming function is still not working. I have read the following articles and it seems that to activate a wireless dimmer, a code is needed in home assistant:
https://community.home-assistant.io/t/deconz-switch-configuration/93346
https://www.home-assistant.io/integrations/deconz/

But my switch is wired and I wonder what to do next to make it works manually?

You advised me the following:
"You can see some return on websocket when you long press the switch ?"
I am a newbie and will need some help to check that (-:

Smanar, 2020-01-11:
Yep the switch in the legrand application do the same job than the option 0101 described in deconz.

It's not hardware because it have already worked with the standard gateway.
It's not a configuration problem, because you can dim the light using deconz.

Can you make a try with deconz turned off (just for 1 minute) ?
I don't know Hass , but I m sure somewhere you can find some logs for the plugin ?

BTW can you give me the firmware version of the device ? you can see it in Phoscon too.

cobch, 2020-01-11:
Ok. I understand. Really strange...
I tried to turn off Deconz but the dimmer still does not react.
The version of the plug is 0029, see below.
[image]

I have managed to check the log, it seems that nothing happen when I press long on the switch manually. I get something only when I switch on or off. See the logs below I get.
[image]

Any ideas?
It seems that I use channel 15, do you think it can be the solution to change it to channel 11 (I saw somewhere that channel 11 should be used)

nassost79, 2020-01-11:
@cobch I 'm using channel 11 but it still doesn't work. I don't think it makes a difference for the issue we 're trying to solve.

cobch, 2020-01-11:
Do you have the same problem?

nassost79, 2020-01-11:
@cobch if we are talking about the lack of dimming control via the actual legrand dimmer switch, yes.

cobch, 2020-01-11:
@nassost79 So we have the same troubles. Really weird! I hope we will managed to fix that because it is nothing I will implement at home otherwise.

Smanar, 2020-01-11:
I will update mine to the same version to make tries.
But it's realy something strange, because the dimming feature is hardware, so I don't care channel or gateway.
Perhaps a new setting.

cobch, 2020-01-11:
@Smanar Thanks. Lets me know if you find something! It will be so great (-:

Smanar, 2020-01-11:
BTW, if you don't have too much device, I think it's a good idea to go on channel 11, even it works today, not sure you won't have problem later, better to do that when you don't have too much devices.

cobch, 2020-01-11:
Ok, should I change it in deconz GUI, and choose 11 instead for 15?
I have tried before to change to 11 and lost some pairing with 3 multi sensor from Xiaomi... But for the moment I have only 15-20 sensors totally so I will follow your advice and repair in case of problem.

Smanar, 2020-01-12:
No, don't worry. I have enought informations.
I have reproduced the problem, I m already on it.

Smanar, 2020-01-12:
Ok, so sorry, It's realy a mystery for me.
I have sniffed all the traffic with the original gateway, no difference with me. I have tried randomly some setting, no changes.

I can't understand how the hardware can be broked, it s supposed to work off line ...

When you long press, the device don't react at all, it don't send command, don't change level, nothing.

cobch, 2020-01-12:
So sad. I have looked on the zigate forum and I wonder if there is some compabilties there? Maybe some help?
https://zigate.fr/le-materiel-compatible-zigate/#inventory_filter

Do you know if it is possible to downgrade the version of the switch to the one you used before which was working with both deconz and manual dimming?

cobch, 2020-01-12:
What is that for? Min-max? Nowhere to set level neither?
[image]

ebaauw, 2020-01-12:
As the header says: that the configuration for attribute reporting.  See section 7.2.3 in the deCONZ user manual, available under the _Help_ menu.

cobch, 2020-01-12:
I will have a look. Thanks. Any ideas about how to activate the dimming function?

Smanar, 2020-01-13:
Yep, I know zigate, I have one, but to test the dimmer we need the last (not public firmware), I have ask help to pipiche and he will probably help us.
So I will have another hardare to make comparaison.

But I still have some other tests to do, I haven't missed a command, but I have perhaps do too much of them, I need to make tries with API disabled or less effective.

And no, we can't downgrade the device, because we haven't the firwmare ^^, and my GF realy don't like the "max power" light, not good for me.

I suspect something in attribute reporting setting too, it' the only command that are not identical.

cobch, 2020-01-13:
Ok. So nice to get information from you. I will wait and sincerely hope it will be fix soon so I can move all my dimmer into Deconz (-:

Smanar, 2020-01-19:
So some news (bad).
I realy don't understnad why the device stop working manualy with deconz (and only it)
I have tried with the API removed, same problem, so it's perhaps from deconz itself, but I can't say If there is something too much or less.
If someone have an idea ...

For the moment I m waiting for a new sniffer.

cobch, 2020-01-20:
So sad to hear that only on Deconz a problem occurs. Shall we adress a special issue to them? Great to hear that you are trying anyway and good luck!

Smanar, 2020-01-24:
Ok, so good news, I have finally found the problem.

I m trying to clean the code, to make a PR, I will be away this WE so need some days.

cobch, 2020-01-29:
Hi Smanar,
Just give us some news when you will have done the work. Do you know whether the wireless dimmer is working or not in Deconz? What about the legrand micromodule?

Smanar, 2020-01-30:
@cobch the code is finish, it is pull request, you can compile it to test it.
For the moment I m waiting validation or comment.

The wireless dimmer is working too, but you need to update the device (like the wired one)
Micromodule is working too.

cobch, 2020-01-30:
@Smanar As mentionned in my first post I am a passionnant newbie but a newbie. I will maybe need some guidance to manage what you describe above (-: Or shall I wait for a new update from Deconz? My GF is joking with me everyday about this dimmer that is not working (-;

Smanar, 2020-01-30:
Looool, I can understand, I had same problem, and I have broked mine just for tests.

To compile my changes:
- you need to be on linux/raspbian.
- you have all the command on githup mainpage "Install deCONZ development package (optional, Linux only)"
- You just need to change the step 1
`git clone https://github.com/Smanar/deconz-rest-plugin.git`
instead of
`git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git`

You can to wait for deconz dev return, but some PR are without answer since 15 days, and not sure they will validate mine.

cobch, 2020-01-30:
Ok, so I have a problem since I have installed Hass.io on a raspberry Pi and so I have not raspbian... I have to pray or is there another solution?

Smanar, 2020-01-30:
Arf, right, will be complicated with Hass.io.

I can send you mine, but it was compiled for raspberry 3, and IDK wich one OS you have on your HASS.Io ? It have lot of chance to don't be compatible.

cobch, 2020-01-30:
Ok. I have HassOs 2.12
I understand if you cannot help me further in that... I asked just in case.

cobch, 2020-02-02:
Hi @Smanar 
I wonder what I can do to inform Deconz to update their program according to your code?
Shall I close this issue or let it open till they update their version?

Thanks and have a nice week end

Smanar, 2020-02-02:
Hello,
It's ok the code was already validated, 2 days ago.
Now we just need to wait for next beta version.

You can keep it open, time for you to test the next version.

cobch, 2020-02-02:
Great. I will wait to close the issue.
Do you know if the wireless switch home/away included in the starting kit is working with Deconz? 

[image]

Smanar, 2020-02-03:
Good question ^^.
I think not, but there is a problem. To include it in deconz, you need to reset it, and if you reset it, it become unusable for the Legrand gateway, and I m sure you know how this device is vital for the Legrand gateway.

So no-one have already tried with the one included in the starting kit (In reality I know someone, but he had problem after).

So to resume, if you have only 1 home/away switch, avoid, it's possible but dangerous.

cobch, 2020-02-03:
I have tried earlier to include it in Deconz but it did not worked. To reconnect  it with Legrand Gateway I just followed the Instructions on #883 
Do not remember where exactly.
It seems I will buy a new wireless switch in order to make my home/away function into home assistant. Safer but more expensive.

Smanar, 2020-02-03:
You need it to enable the permit join on Legrand gateway (pressing the "middle" button), so how have you re-include it ?

The wireless switch is a little cheaper than the home/away and have been already tested in deconz.

cobch, 2020-02-03:
Hi @Smanar 
I just follow the instructions given by Legrand to @bsheepdiy:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/883#issuecomment-570767677

But as you mentioned earlier, it is better to buy a new wireless switch and use it as a "Home/Away" switch since I can soon connect it to my Home assistant and call some script with it (-:

bsheepdiy, 2020-02-04:
@cobch 
I've try modified deconz to include/whitelist "Home/Away" device and it doesn't pair ...
Using wireshark dump, the device only send beacon and that's all. So I ve tried using zigbee2mqtt (version 1.9.0) I have the same issue. I think the Home/away packaged with the "Prise control" has a special firmware which is locked to the Legrand official network.

I'll need to buy an additional "Home/Away" device which will be unlocked ...

cobch, 2020-02-04:
@bsheepdiy same issue for me when I tried!

Smanar, 2020-02-04:
It's perhaps something missing in code.
The remote do "read attribute request" on cluster 0000 attribute F000 (to other device/gateway) ?
If yes, the problem is from here.

And no, I dont think legrand have locked devices, they have realy try to respect the zigbee protocol. And they don't make special device, if for exemple you loose the "special remote", you can buy one other alone, you don't need to buy a complete kit.

cobch, 2020-02-04:
In my case I think I will need to spend an extra 50 euros to use the Home/away command within Home assistant but it is still frustrated not to be abble to integrate it into Deconz. Same thing for the smart socket with diamond...

bsheepdiy, 2020-02-04:
@cobch 
Wait, the Home/away device is not yet supported by deconz ...

cobch, 2020-02-04:
@bsheepdiy I will buy instead a wireless switch since that one is supported. Do you know if the smart outlet with diamond (gateway) will be supported one day?

bsheepdiy, 2020-02-04:
@cobch 
I dont think you can separate the gateway from the device. The reset buton is the same ...

Smanar, 2020-02-04:
> 
> 
> @cobch
> I dont think you can separate the gateway from the device. The reset buton is the same ...

Yep, I have already try to see if we can separate them, but it's realy a device make to work with the diamond.

And I m agree Legrand is realy expensive (the minimum I have found for a simple wireless switch on celiance is 41 Euros), but for the moment it's realy good devices, less battery consumption, they don't leave the network like xiaomi. I have some Legrand device, and I m realy happy with them.

@cobch you can't transform an existing switch in a double one ? and using one of them with a cheaper xiaomi device (with some hack) ?

Pipiche too have tried but this device is a little special

> The Home/Away remote which is provided with the Legrand Hub and which in addition of Home/Away scene allow to switch the Legrand Hub in pairing mode, do not show any Zigbee traffic .

cobch, 2020-02-04:
@ Smanar I agree Legrand is great quality (except the plastic that breaks that you have to remove when you want to pair it to deconz) and I have already equiped all my flat with wired switches. I just think Legrand should consider to decrease the price of their wireless switch since it is the same price as a wired one. Strange. I will wait the new release of Deconz and order a new switch as soon I have tested that is is working.  Great thanks for @ Smanar that really help/make a great job!

Smanar, 2020-02-04:
Lol, yep a 5mm wireless switch  cost the same price than a 4cm  one working with 230V ^^.

And I have too broken one too, I have one of them with a little hole make with a drill :) , because I m using the same device for production and tests , so it's easier for reset it.

cobch, 2020-02-15:
Hi again @Smanar 
I am still waiting for the new update of Deconz to solve my dimmer problem. Do you have any ideas how long will I wait?

Another question. I am now playing with the wireless dimmer from Legrand (-: I bought a new one (to avoid to play with my current home settings) and its version is 25. I do not know how to update it (the current version showing for my other wireless dimmer is 33). Any ideas how to force the update of the firmware version thought Legrand gateway?

Smanar, 2020-02-15:
The new beta version is just out http://deconz.dresden-elektronik.de/ubuntu/beta/ for ubuntu

Easy to force the update, you just need to let your device on Legrand network for some hours, I think more for wireless device, but the procedure is the same.

And for information @bsheepdiy have discoverer some tips for the home/away switch.

cobch, 2020-02-15:
Ok. I run Home Assistant and deconz on raspberry pi so I will wait a bit longer (-;
I will wait tomorrow and hopefully the firmware will be updated.

I will have a look for the home/away switch since I am interested by it as well. Thanks for the information.

Smanar, 2020-02-15:
You have too the version for raspbian > http://deconz.dresden-elektronik.de/raspbian/beta/

cobch, 2020-02-16:
Hi again @Smanar 
I am now trying to add my wireless switch to deconz. I reset the switch with 8s long press till the red light shows and then press each second till the blue light is showing. The switch does not appear in Phoscon app but appears in Deconz WebUI. 
In Home assistant I only get an entitiy showing the battery level of my switch. It is not possible to use it as a switch since I guess I am missing some installation... 
What to do next to make the switch showing in Phoscon and to make it usable in Home assistant?

cobch, 2020-02-16:
I added the wireless switch with the "Add new lights" in Phoscon. Is it right or should I include it as a "new switch" instead?

cobch, 2020-02-16:
[image]

[19 more replies]

closed 2020-02-18 by cobch