dresden-elektronik/deconz-rest-plugin #2310: Deconz config folder and mounts for synology nas

IIIdefconIII, 2020-01-11:
Hi, first of all, i love this deconz package.

Im stuck with a question, whenever i update my deconz image from synology it removes my settings,
Previous this was for al docker containers for synology.
I  have SMB share called docker, withing that i have several folders from several docker containers which saved its settings in that folder. I will show you a example down below from Radarr. So whenever i update Radarr to a newer image on synology and i "clear" the docker cache, im getting the same settings because of the settings i have in the docker container. Like this:
[image]

The folder that inclde the settings are mounted a a folder on my NAS:

Now i tried several settings for the deconz container, as i updated it and cleared the cache i lost my whole  config.

[image]

When i ssh in to my nas i simply cant find any docker folder that is related with the settings of deconz. Maybe is obvious set somewhere but i cannot find it.

Which setting should/can  i use to save the dconz settings to a mounted folder.

This would help me alot. Thank you in advance and love youre project, is there any play i can buy you a coffee/beer?

kTitan, 2020-01-15:
If you use the docker image from https://hub.docker.com/r/marthoc/deconz/ you should need only the volume mount for /root/.local/share/dresden-elektronik/deCONZ
Have a look in its Readme, there is a example for the volume mount.

IIIdefconIII, 2020-01-16:
Thank you, indeed now its saving to my path.
[image]
[image]

Case closed added some screenshots if anyone ever needs it

closed 2020-01-16 by IIIdefconIII