dresden-elektronik/deconz-rest-plugin #2311: node-red/deconz websocket permanent reload

pandschai, 2020-01-11:
Hi everybody,
I hope this is not to far of any reasonable discussion - i am quite new to all this...
My Problem:
Installed Deconz, installed Node-Red on a Raspberry Pi 4B with Buster...
Got some Aqara door senors which got recognized by Deconz (and the web-app).
Installed: npm install node-red-contrib-deconz in ".node-red"
Started node-red
installed the deconz-node and the sensor was recognized but...
after a few seconds the webocked is interrupted and automatically restarted. During this time the sensor deconnects. This is what is wirten in the terminal:

11 Jan 21:01:25 - [info] Flows starten
11 Jan 21:01:25 - [error] [deconz-server:3a5a8754.6fc1d8] TypeError [ERR_INVALID_URL]: Invalid URL: ws://:
11 Jan 21:01:25 - [info] Flows gestartet
11 Jan 21:01:25 - [info] [sqlitedb:88b729ea.b4bfe8] opened /home/pi/xxx.db ok
11 Jan 21:01:25 - [info] [deconz-server:Conbee2] WebSocket opened
11 Jan 21:01:25 - [info] [deconz-server:Conbee2] WebSocket opened
11 Jan 21:01:25 - [info] Server wird jetzt auf http://XXX.X.X.X:1880/ ausgeführt.
11 Jan 21:01:35 - [red] Uncaught Exception:
11 Jan 21:01:35 - TypeError [ERR_INVALID_URL]: Invalid URL: ws://:
    at onParseError (internal/url.js:241:17)
    at new URL (internal/url.js:319:5)
    at initAsClient (/home/pi/.node-red/node_modules/ws/lib/websocket.js:477:17)
    at new WebSocket (/home/pi/.node-red/node_modules/ws/lib/websocket.js:72:7)
    at DeconzSocket.connect (/home/pi/.node-red/node_modules/node-red-contrib-deconz/lib/deconz-socket.js:50:27)
    at Timeout.setTimeout [as _onTimeout] (/home/pi/.node-red/node_modules/node-red-contrib-deconz/lib/deconz-socket.js:111:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)

Tried to figure out how to fix the "[error] [deconz-server:3a5a8754.6fc1d8] TypeError [ERR_INVALID_URL]: Invalid URL: ws://:" but couldn't find anything.

Any idea how to fix this websocket? Thanx!

andreypopov, 2020-01-13:
can you attach pic of your server node settings? 
need more info: node-red version, plugin version, do you use docker ?

pandschai, 2020-01-13:
hi andrey thx for answering, 

Node-RED Version: v1.0.3
Node.js  Version: v10.18.0
deConz connecitivity nodes for nodered (node-red-contrib-deconz version): 1.1.10
I use ConbeeII as gateway

Screenshot Server Node 1:
[image]

Screenshot server node 2:

[image]

And here again the complete terminal window which repeats over and over again:

3 Jan 17:23:30 - [info] Flows starten
13 Jan 17:23:30 - [error] [deconz-server:3a5a8754.6fc1d8] TypeError [ERR_INVALID_URL]: Invalid URL: ws://:
13 Jan 17:23:30 - [info] Flows gestartet
13 Jan 17:23:30 - [info] [sqlitedb:88b729ea.b4bfe8] opened /home/pi/zeitschriftenmessung_zlb.db ok
13 Jan 17:23:30 - [info] [deconz-server:Conbee2] WebSocket opened
13 Jan 17:23:30 - [info] [deconz-server:Conbee2] WebSocket opened
13 Jan 17:23:30 - [info] Server wird jetzt auf http://XXX.X.X.X:1880/ ausgeführt.
13 Jan 17:23:40 - [red] Uncaught Exception:
13 Jan 17:23:40 - TypeError [ERR_INVALID_URL]: Invalid URL: ws://:
    at onParseError (internal/url.js:241:17)
    at new URL (internal/url.js:319:5)
    at initAsClient (/home/pi/.node-red/node_modules/ws/lib/websocket.js:477:17)
    at new WebSocket (/home/pi/.node-red/node_modules/ws/lib/websocket.js:72:7)
    at DeconzSocket.connect (/home/pi/.node-red/node_modules/node-red-contrib-deconz/lib/deconz-socket.js:50:27)
    at Timeout.setTimeout [as _onTimeout] (/home/pi/.node-red/node_modules/node-red-contrib-deconz/lib/deconz-socket.js:111:35)
    at ontimeout (timers.js:436:11)
    at tryOnTimeout (timers.js:300:5)
    at listOnTimeout (timers.js:263:5)
    at Timer.processTimers (timers.js:223:10)
nodered.service: Main process exited, code=exited, status=1/FAILURE
nodered.service: Failed with result 'exit-code'.
nodered.service: Service RestartSec=100ms expired, scheduling restart.
nodered.service: Scheduled restart job, restart counter is at 174.
Stopped Node-RED graphical event wiring tool.
Started Node-RED graphical event wiring tool.
13 Jan 17:23:42 - [info]
Willkommen bei Node-RED!
===================
13 Jan 17:23:42 - [info] Node-RED Version: v1.0.3
13 Jan 17:23:42 - [info] Node.js  Version: v10.18.0
13 Jan 17:23:42 - [info] Linux 4.19.75-v7l+ arm LE
13 Jan 17:23:42 - [info] Paletten-Nodes werden geladen
13 Jan 17:23:44 - [info] Einstellungsdatei: /home/pi/.node-red/settings.js
13 Jan 17:23:44 - [info] Kontextspeicher: 'default' [ module=memory]
13 Jan 17:23:44 - [info] Benutzerverzeichnis: /home/pi/.node-red
13 Jan 17:23:44 - [warn] Projekte inaktiviert: editorTheme.projects.enabled=false
13 Jan 17:23:44 - [info] Flow-Datei: /home/pi/.node-red/flows_raspberry.json
13 Jan 17:23:44 - [warn]
---------------------------------------------------------------------
Die Datei mit den Datenflowberechtigungsnachweisen wird mit einem vom System generierten Schlüssel verschlüsselt.
Wenn der vom System generierte Schlüssel aus irgendeinem Grund verloren geht, werden Ihre Berechtigungsnachweise
Die Datei kann nicht wiederhergestellt werden. Sie müssen sie löschen und erneut eingeben.
Ihre Berechtigungsnachweise.
Sie sollten Ihren eigenen Schlüssel mit Hilfe der Option 'credentialSecret' in
Ihre Einstellungsdatei. Node-RED wird dann Ihre Berechtigungsnachweise erneut verschlüsseln.
Datei mit dem ausgewählten Schlüssel beim nächsten Deployen einer Änderung verwenden.
---------------------------------------------------------------------
13 Jan 17:23:44 - [info] Flows starten
13 Jan 17:23:44 - [error] [deconz-server:3a5a8754.6fc1d8] TypeError [ERR_INVALID_URL]: Invalid URL: ws://:
13 Jan 17:23:44 - [info] Flows gestartet

closed 2020-05-20 by stale[bot]

TiagoJVO, 2020-11-11:
I had the same problem, basically somehow I end-up configuring a deconz server without filling the fields, and it got deployed... I went to the filesystem and manually changed the `\config\node-red\flows.json` file and deleted the server config for the `deconz-server` (you can just ctrl-F and delete the json object with that type)