dresden-elektronik/deconz-rest-plugin #2313: Add new device to "Supported devices list" (Samotech SM308)

ottelo9, 2020-01-12:
I bought two of these zigbee actors (in-wall-switch?):

[Product homepage Samotech SM308](https://www.samotech.co.uk/products/zigbee-ac-on-off-switch-compatible-with-philips-hue-amazon-echo-plus-and-echo-show-2nd-gen/)

They are working fine with Deconz! Please add it to the supported device list.

greencharlie100, 2020-01-14:
Do the dimmers work file with Deconz? Product link below.

SM309 ZIGBEE DIMMER SWITCH

Smanar, 2020-01-14:
Juts for information the page https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Supported-Devices is free to edit, if you hve a github account you can edit it yoursel without problem.

ottelo9, 2020-01-14:
> Do the dimmers work file with Deconz? Product link below.
> 
> SM309 ZIGBEE DIMMER SWITCH
> https://www.samotech.co.uk/products/zigbee-dimmer-switch-hue-compatible/

Sorry but I only have the SM308 actors. 

@Smanar 
Ah thanks for that hint.

closed 2020-01-14 by ottelo9