dresden-elektronik/deconz-rest-plugin #2316: Compilation error

Cortex33, 2020-01-13:
Hello,

I tried so many time to compile the api in order to add it to my home assistant deconz 5.1 add on.
And I'm stuck with the error :
de_web_plugin.h:16:10: fatal error: deconz.h: Aucun fichier ou dossier de ce type
 #include <deconz.h>
          ^~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:517: release/bindings.o] Error 1
make[1]: *** Attente des tâches non terminées....
In file included from authorisation.cpp:13:
de_web_plugin_private.h:23:10: fatal error: sqlite3.h: Aucun fichier ou dossier de ce type
 #include <sqlite3.h>
          ^~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:499: release/authorisation.o] Error 1
make[1] : on quitte le répertoire « deconz-rest-plugin »
make: *** [Makefile:40: release] Error 2

May it possible to have some help ? I need to add this to my installation in order to add Heiman sensor to ConBeeII.

In advance, thanks you.

Smanar, 2020-01-13:
Have you installed the developpement package ? sudo apt install deconz-dev

Cortex33, 2020-01-13:
Hum no :1st_place_medal: , and now strangely it seem to work fine ... Thanks you so much.
I close my issue here.

closed 2020-01-13 by Cortex33

oivindoh, 2020-02-01:
@Smanar is there a development package for arm64 somewhere? I'm trying to build master for arm64, but I can't find a compatible dev package to install.