dresden-elektronik/deconz-rest-plugin #2317: Add Xiaomi Mi Flora Flower Sensor as Bluetooth accessory

patrickscheips, 2020-01-13:
Hi all,

I own a Xiaomi Mi Flora Flower Sensor that uses Bluetooth Low Energy to expose a plant's fertitlity level and more data. There is a [Homebridge plugin](https://www.npmjs.com/package/homebridge-mi-flower-care) that allows to use this sensor with home bridge.

How do I now expose this sensor to homebridge? homebridge-hue seems to get the accessories, that are normally added to the homebridge config.json, from Deconz. Can I somehow add the sensor to Deconz or is there a way to bypass Deconz and add it to homebridge manually? From what I understand, Deconz only supports Zigbee devices and not Bluetooth devices — so I guess I will have to bypass Deconz somehow.

I hope you guys can help me with this.

Thanks,
Patrick

ebaauw, 2020-01-13:
In a standalone homebridge installation, you would add the other plugin to your config.json, and run homebridge with two plugins.  Neither deCONZ nor homebridge-hue would need to know about that.  You might want to double check that the deCONZ image won't overwrite your changes, but otherwise you would be OK.

Having said that, I do see some issues particularly for homebridge-mi-flower-care:
- Note that Bluetooth and ZigBee (as well as Wifi) share the same 2.4 GHz band.  I'm not sure if I would want to run a Bluetooth application on the same host as deCONZ.
- To access the Bluetooth hardware, homebridge-mi-flower-care requires homebridge to run under `root` privileges.  I definitely don't recommend that.

patrickscheips, 2020-01-15:
Thanks for your reply and for clarifying the disadvantages of this approach. For now, I will refrain from using the Flower Sensors with homebridge because of these potential issues.

closed 2020-05-22 by stale[bot]