dresden-elektronik/deconz-rest-plugin #2320: Right dot not showing for Spirit Zigbee Thermostat + Rasperry Pi 4

piepor, 2020-01-14:
Hi,
I'm using Conbee II with Rasperry PI 4. I have a Spirit Zigbee Thermostat: i first tried it with a pc and it was all ok, I could control the set point etc.. The only strange thing was the green dot still blinking. When I installed it on the Rasperry PI 4, the green dot was still blinking but now the right dot (the one needed in order to control the device) didn't show up. I tried leaving and rejoin the network and to reset/repair the device but nothing worked. Has anyone some idea about this strange behaviour?

ebaauw, 2020-01-15:
deCONZ hasn’t read the _Simple descriptor(s)_.   You can do so manually from the left dot.  I’ve seen some intermittent issue that deCONZ wouldn’t read the descriptor while the network is open, but I haven’t been able to put my hand on it.

After the dot appears, search for new devices from Phoscon and read the attributes from the _Basic_ cluster to make the REST API plugin create the resource.

piepor, 2020-01-15:
> deCONZ hasn’t read the _Simple descriptor(s)_. You can do so manually from the left dot. I’ve seen some intermittent issue that deCONZ wouldn’t read the descriptor while the network is open, but I haven’t been able to put my hand on it.

I tried manualiy from the left dot but when I select Read Simple Descriptor(s), the box uncheck immediately and nothing happened. This time the left dot is most of the time off with a blue blinking occasionally. I think network is closed: when I press "Edit/Permit Join" is written 0. Is there another way to check if the network is opened?

> After the dot appears, search for new devices from Phoscon and read the attributes from the _Basic_ cluster to make the REST API plugin create the resource.

Do you mean Phoscon WebApp? I only have the possibility to add lights, switches and sensors. How can I search for a new, general device?

closed 2020-03-19 by piepor