dresden-elektronik/deconz-rest-plugin #2321: Support for Linkind 1 key Remote Control / ZS23000178

madbrain-, 2020-01-15:
Hi,
thank you for this product, I really love it. Works great until now. 
How can I add this device to Phoscon? It gets recognized in deconz:

Model ZS23000178
FCC ID 2ASHKZS23000178
https://www.amazon.de/dp/B07V53MLNT/ref=cm_sw_em_r_mt_dp_U_x3WhEbP0WFG06
bought it for about 10 € a few weeks ago

[image]
[image]
[image]

Here is a Log, that may be related to a push on the button:
22:38:59:921 APS-DATA.indication srcAddr: 0x04f8, srcEp: 0x01 dstAddrMode: 1, profile: 0x0104, cluster: 0x0006, lqi: 151, rssi: -72

But if I use the phoscon-app and search for new switches it doesn't get recognized. I assume, it has to be added to the API or soething like this?
Maybe I could help. I'm very interested in the Zigbee Bridge, maybe I could help to integrate the switch.

Let me know, if you need more information!

Thanks!
regards, Thomas

madbrain-, 2020-04-08:
I read somewhere, that if the search in Phoscon is active and the basic cluster gets read, some more Information are show. I did that with this Button and the output of the basic cluster changed:

[image]

Can somebody point me in the right direction, what has to be done to get this Button in phoscon / iobroker? Maybe I can do that on my own, but I can't find any documentation how to add a new device.

Thanks!
regards, Thomas

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

madbrain-, 2020-06-09:
Hi @Mimiix,
it is not solved as I did never get an answer here. How can I help to integrate this switch?

Mimiix, 2020-06-09:
@madbrain-  i can ;)

@SwoopX Is there anything that is missing?

SwoopX, 2020-06-09:
Power configuration cluster would be helpful as well. Later on, we'd require some additional debug output from deconz for finetuning...

madbrain-, 2020-06-09:
Here is the power configuration cluster:

[image]

Thank you for your help!

madbrain-, 2020-07-01:
What can I do to help with the integration? Can I find somewhere a developer documentation?

OrbitingOcelot, 2020-07-29:
I would also be interested in seeing support for this.

SwoopX, 2020-07-29:
Can someone compile the rest plugin and test?

madbrain-, 2020-07-30:
I can try it on the weekend. Great to hear some news! Thank you very much! :-)

madbrain-, 2020-08-03:
I tried to compile the plugin. I did the following:
sudo apt install deconz-dev
git clone https://github.com/dresden-elektronik/deconz-rest-plugin.git (master)
qmake && make -j2
cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
reboot

I assume the plugin got loaded? How can I check that?

But I can't see the device in the Phoscon App nor in the rest api.

SwoopX, 2020-08-03:
> I assume the plugin got loaded? How can I check that?

I assume your deconz is still working, that's the check since you've overwritten your previously installed version of the plugin.

> But I can't see the device in the Phoscon App nor in the rest api.

Yes, because it is not supported. I just asked if you can do it, not to do it 😉

madbrain-, 2020-08-04:
> I assume your deconz is still working, that's the check since you've overwritten your previously installed version of the plugin.
Yes, everything is working, as far as I can notice until now.

> Yes, because it is not supported. I just asked if you can do it, not to do it wink
okay :-)

SwoopX, 2020-08-08:
Hey @madbrain- 

Haven't forgotten about you, but was busy to finish a new feature. So, I got some code prepared for you to test now, but it will require some doings on your end.

Here are the steps to download and compile my version:

```
sudo systemctl stop deconz
sudo systemctl stop deconz-gui
git clone --branch linkind https://github.com/SwoopX/deconz-rest-plugin.git linkind
cd linking
qmake && make -j2
cp button_maps.json .local/share/dresden-elektronik/deCONZ/
sudo cp /usr/share/deCONZ/plugins/libde_rest_plugin.so /usr/share/deCONZ
sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins
/usr/bin/deCONZ --dbg-info=2 > debug
```

Then, start Phoscon sensor search and pair the button. Make sure to has been properly reset before. When the device has paired, do all button push variations the device should support. Note their order and ideally, what the button press should have done, if known.

madbrain-, 2020-08-09:
Thank you @SwoopX ,
Great to hear some progress, thank you very much!
I will try that immediately after I got my new disk. I had a disk failure on my pi and I am waiting for thew new one.

madbrain-, 2020-08-24:
Hi thank you all for the work. My deconz is working again. The instruction manual of the Button says it has on/off/dimming:
"press the button to turn the light on or off. Keep pressing to dim the light to max/min brightness smoothly"

SwoopX, 2020-08-30:
@madbrain- @mralex07 This here implies that there's more commands the device can issue:

> Keep pressing to dim the light to max/min brightness smoothly

As per feedback provided here: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2321#issuecomment-678463924 , the debug output translates into the following:

**Off command** - no button map for: ZBT-DIMSwitch-D0001 ep: 0x01 cl: 0x0006 cmd: 0x00 pl[0]: 000 -- first short click on button
**On command** - no button map for: ZBT-DIMSwitch-D0001 ep: 0x01 cl: 0x0006 cmd: 0x01 pl[0]: 000 -- second short click on button
**Move up command** - no button map for: ZBT-DIMSwitch-D0001 ep: 0x01 cl: 0x0008 cmd: 0x01 pl[0]: 000 -- long press on button
**Stop command** - no button map for: ZBT-DIMSwitch-D0001 ep: 0x01 cl: 0x0008 cmd: 0x03 pl[0]: 000 -- release button

In that sense, I'd at least expect to see in addition:

ep: 0x01 cl: 0x0008 cmd: 0x01 pl[0]: 001 -> **Move down command**
ep: 0x01 cl: 0x0008 cmd: 0x03 pl[0]: 001 -> **Stop command for move down**

Can you guys please double check the logs and try to force the move down? Maybe the manual gives hints on that. Please also take double press + hold into consideration.

mralex07, 2020-08-31:
Hi, @SwoopX 
You are right, i`ve checked logs, here its are:
ep: 0x01 cl: 0x0008 cmd: 0x01 pl[0]: 001          -> first long press start                                       
ep: 0x01 cl: 0x0008 cmd: 0x03 pl[0]: 000         -> release                                        
ep: 0x01 cl: 0x0008 cmd: 0x01 pl[0]: 000         -> second long press start                                        
ep: 0x01 cl: 0x0008 cmd: 0x03 pl[0]: 000        -> release

double press:
ep: 0x01 cl: 0x0006 cmd: 0x00 pl[0]: 000                                                 
ep: 0x01 cl: 0x0006 cmd: 0x01 pl[0]: 000

double press + hold:
ep: 0x01 cl: 0x0006 cmd: 0x01 pl[0]: 000                                                 
ep: 0x01 cl: 0x0006 cmd: 0x00 pl[0]: 000                                                 
ep: 0x01 cl: 0x0008 cmd: 0x01 pl[0]: 000                                                 
ep: 0x01 cl: 0x0008 cmd: 0x03 pl[0]: 000

closed 2020-09-15 by manup

mathieg2, 2021-01-15:
Hi - I have this device, It is detected in phoscon, but I'm unable to assign any actions to it - it seems I can only get it's battery level. Pressing the button does generate activity in debug log:

/usr/bin/deCONZ --dbg-info=2 >debug.log

Is there some instructions I can look up to find out if I am doing something wrong? I'm on version 2.09.

Graeme

Mimiix, 2021-01-15:
@mathieg2 phoscon needs to add support for it too.

You can use it with the rest api

ebaauw, 2021-03-13:
@SwoopX, I don't understand: this is a one-button device, but deCONZ exposes two buttons for it?!

SwoopX, 2021-03-22:
@ebaauw 

What should it be then? Twice short, long and hold on button 1?

```
"linkind1keyMap": {
            "vendor": "Linkind",
            "doc": "1 button remote ZS232000178",
            "modelids": ["ZBT-DIMSwitch-D0001"],
            "map": [
                [1, "0x01", "ONOFF", "ON", "0", "S_BUTTON_1", "S_BUTTON_ACTION_SHORT_RELEASED", "On"],
                [1, "0x01", "ONOFF", "OFF", "0", "S_BUTTON_2", "S_BUTTON_ACTION_SHORT_RELEASED", "Off"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "0", "S_BUTTON_1", "S_BUTTON_ACTION_HOLD", "Move up"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_1", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"],
                [1, "0x01", "LEVEL_CONTROL", "MOVE", "1", "S_BUTTON_2", "S_BUTTON_ACTION_HOLD", "Move down"],
                [1, "0x01", "LEVEL_CONTROL", "STOP", "0", "S_BUTTON_2", "S_BUTTON_ACTION_LONG_RELEASED", "Stop"]
            ]
        },
```