dresden-elektronik/deconz-rest-plugin #2322: Philips Hue Presence sensor - state doesnt change from "detected"

JamieP83, 2020-01-15:
hey all,

Conbee II
version: 2.05.72 (12/12/2019)
firmware: 264A0700

Device: philips hue Motion Sensor

Issue: the sensor will change to "motion = detected" however, it never changes back again?
it just stays "on"

I know its working overall, as the LUX will change, so the device is working overall.

ebaauw, 2020-01-15:
The sensor should send a message when it no longer detects motion, which the REST API plugin picks up to reset `state.presence`.  What is the sensor's `config.delay` value?

JamieP83, 2020-01-15:
Sorry, how do I check that? I have this via home assistant

ebaauw, 2020-01-15:
Thru the API.

JamieP83, 2020-01-16:
Sorry, I’m unclear where you mean im afraid 

Is there someone I can find out more on finding API info?

JamieP83, 2020-01-19:
i cant find "config.delay" in my home assistant logs?

ebaauw, 2020-01-19:
The API is documented here: http://dresden-elektronik.github.io/deconz-rest-doc/.  The deCONZ API is modelled after the Hue API; you might want to check out the Hue API as well: https://developers.meethue.com/develop/get-started-2/.

To check `config.delay`, you need to do a GET of the `/sensors` resource for the Hue motion sensor.  I don't know home assistant an wouldn't know whether it logs this attrobute.

JamieP83, 2020-01-28:
hey,
sorry, i dont seem to be able to run this. 

if i run the GET http://dresden-light.appspot.com/discover
that works with this result
```
[
    {
        "id": "00212EFFFF05073F",
        "internalipaddress": "172.30.33.0",
        "macaddress": "00212EFFFF05073F",
        "internalport": 40850,
        "name": "PhosconGWconbee2",
```

however i run then
GET: https://172.30.33.0:40850/sensors 
but nothing?
"could not get any response"

ebaauw, 2020-01-28:
You need to create an api key (username) by doing a POST to `https://172.30.33.0:40850/api/` and then GET `https://172.30.33.0:40850/api/`_username_`/sensors`.  Make sure to unlock the gateway before the POST, or you'll get a "linkbutton not pressed" error.

JamieP83, 2020-01-28:
thanks - i keep trying that but i cant get the POST for the API key to work

[image]

[image]

JamieP83, 2020-01-28:
ah, ive looked in the logs.

the delay is set to 1202? is that in seconds? or ms?

>    '7': {'config': {'alert': 'none',
>                               'battery': 100,
>                               'delay': 1202,
>                               'ledindication': False,
>                               'on': True,
>                               'pending': [],
>                               'reachable': True,
>                               'sensitivity': 2,
>                               'sensitivitymax': 2,
>                               'usertest': False},
>                    'ep': 2,
>                    'etag': '1f0d135e4c7bc074090166ea9fbac900',
>                    'manufacturername': 'Philips',
>                    'modelid': 'SML001',
>                    'name': 'Motion Sensor',
>                    'state': {'lastupdated': '2020-01-28T21:29:40',
>                              'presence': False},
>                    'swversion': '6.1.1.27575',
>                    'type': 'ZHAPresence',
>                    'uniqueid': '00:17:88:01:03:2b:f1:e8-02-0406'},

ebaauw, 2020-01-28:
Seconds, so that's 20 minutes.  Change it by a POST to `https://172.30.33.0:40850/api/`_username_`/sensors/7/config` with a body of `{"delay": 0}`.

> thanks - i keep trying that but i cant get the POST for the API key to work

Are you sure deCONZ is running on that ip address and port?  Does a GET of `https://172.30.33.0:40850/api/config` return anything?

JamieP83, 2020-01-28:
So GET: https://172.30.33.0:40850/api/config
doesnt do anything? 
[image]

however GET: http://dresden-light.appspot.com/discover
gives the following:
[image]

JamieP83, 2020-01-28:
ive worked it out!
thanks for the help.

so now, when i run.
GET: hassio.local:40850/api/config

```
{
    "apiversion": "1.16.0",
    "bridgeid": "00212EFFFF05073F",
    "datastoreversion": "60",
    "devicename": "ConBee II",
    "factorynew": false,
    "mac": "02:42:ac:1e:21:00",
    "modelid": "deCONZ",
    "name": "PhosconGWconbee2",
    "replacesbridgeid": null,
    "starterkitid": "",
    "swversion": "2.5.72"
}
```
[image]

JamieP83, 2020-01-28:
when i try and run the update, i get the following error?

POST: hassio.local:40850/api/xxx/sensors/7/config
Body: {"delay": 0}

[image]

JamieP83, 2020-01-28:
ah!
its "put" not "post" 
seems to work now :)

thanks for all the help! 
issue can be closed down

ebaauw, 2020-01-28:
Yes, PUT.  My bad.

closed 2020-06-05 by stale[bot]