dresden-elektronik/deconz-rest-plugin #2323: Heiman HS1SA-N problem

Sjonnie2018, 2020-01-15:
Hi All,

I am having a hard time getting my Heiman HS1SA-N smoke detectors to show up in the Phoscon app and get them properly configured in Domoticz.

Some basic info:
ConBee II: latest firmware on Raspberry Pi 3B + (264A0700)
Deconz: latest version (2.05.72)
Domoticz: latest stable (4.10717)
Domoticz DeConz plugin from @Smanar : latest version (1.0.11)

I can add the sensor in the Phoscon app, it gets detected but after I press "ready" to confirm the pairing the device does not show up. It is however visible in DeConz gui on my RPi3. The DeConz Plugin for Domoticz parses the sensors to Domoticz but there is shows the devices as On/Off switch. Which is to be expected for unknown devices.

The information I get from DeConz:
[url=https://postimg.cc/ykwnRDzH][img]https://i.postimg.cc/ykwnRDzH/Deconz-settings-Heiman-HS1-SA-N.png[/img][/url]

This is what I see in Domoticz:
[url=https://postimg.cc/0b6crpFn][img]https://i.postimg.cc/0b6crpFn/Heiman-Smoke-sensors-in-Domoticz.png[/img][/url]

I have to coding skills of a Kangaroo. I hope you can help me and explain to me how to get this to work in "noob" speak ;) I have looked at issue #1711 but I can't make heads or tails of it :(

TIA

Greetz,

Sjonnie

SwoopX, 2020-01-15:
Can you please also post a screenshot of the basic cluster while having read the attributes?

Sjonnie2018, 2020-01-16:
Hi SwoopX,

Tx for you reply! I have added a screenshot of the basic cluster of the smoke sensor. Weird values :?

[image]

SwoopX, 2020-01-17:
Strange indeed. Have you tried reading it multiple times? Eventually, this does not always work on the first time. Without that data, there's not too much we can do here...

ebaauw, 2020-01-17:
You probably need to press the test button when reading the attributes, to wake the device.  Be sure to cover your ears.

SCoe2k, 2020-01-18:
I have the same sensor, but cannot connect at all via phoscon web gui. Pressed the button with the needle for 5 secs. Green led is blinking, but will not be seen by phoscon web. Hardware conbee2 stick.
Is this maybe a new hardware release?

Any clue?

SCoe2k, 2020-01-18:
[image]

Sjonnie2018, 2020-01-18:
If you press the button for 5 seconds you reset the devices networksettings. Press the button for no more than 2 seconds.

By the way: your stick is not on the latest firmware I believe and the latest version of the software is 72

⁣BlueMail voor Android downloaden ​

Sjonnie2018, 2020-01-18:
OK, if I press the test button, something happens alright. Not sure if it is enough (sensor is not available in Phoscon). A few data is greyed out. That happened when I pressed the read button and the test button on the device at the same time.

The screenshots:
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>
[image]<br/><br/>

ebaauw, 2020-01-18:
> A few data is greyed out.

That's normal, these are attributes that the sensor simply doesn't support.  The greying happens, because the sensor reports back _unsupported attribute_, so it's awake alright.

> sensor is not available in Phoscon

Phoscon is "just" another REST API client.  Like any other client, it needs to be updated to support new devices/device types, after support has been added to the REST API.  Note that Phoscon is _not_ open source, so only dresden elektronik can add support for new devices in Phoscon.

Sjonnie2018, 2020-01-18:
Tx for the explanation. I don't really mind the sensor is not available in the Phoscon app. I only use that to check devices after setting them up. All other settings, scenes and scripts are stored in Domoticz. If it's possible to have it show up there as a smoke sensor I am more than happy :)

SCoe2k, 2020-01-18:
> If you press the button for 5 seconds you reset the devices networksettings. Press the button for no more than 2 seconds. By the way: your stick is not on the latest firmware I believe and the [...]

Thanks for the 2s vs 5s. 
Firmware is as per my Knowledge uptodate

SCoe2k, 2020-01-18:
> That's normal, these are attributes that the sensor simply doesn't support. The greying happens, because the sensor reports back _unsupported attribute_, so it's awake alright.
>
>
> Phoscon is "just" [...]

Thank you. Makes Sense. Did Not think about seeing phoscon as rest-API Client. That explains some things.

Sjonnie2018, 2020-01-18:
[image]
Might be up to date but I am not sure ;)

SwoopX, 2020-01-18:
So, is it working? Also noticed we have a new jennic MAC for Heiman here...

SCoe2k, 2020-01-19:
No. Just tried it this morning with 2s. Does not connect to network. Maybe I am doing something wrong. Up to now I only connect via phoscon app. Is there any other way to connect a new end-device?
@Sjonnie2018 : Strange Firmware topic. A0 is the highest number, but 90 has the latest timestamp

Sjonnie2018, 2020-01-19:
Unfortunately no. The device is still "translated" by @Smanar plugin as a switch so that means it is not recognized. Funny thing is that after the read sessions the battery status of the sensor does show up in Domoticz.

Is here anything I can do? I have three of these little gems and plan to have them switch on all lights in case of an alert and send a message to my cell. I can only do that in Domoticz so I am eager to get it working.

TIA

Sjonnie2018, 2020-01-19:
> * jennic MAC for Heiman*

Sorry for my ignorance but what's a jennic MAC? Especially the jennic part ;)

Sjonnie2018, 2020-01-21:
> No. Just tried it this morning with 2s. Does not connect to network. Maybe I am doing something wrong. Up to now I only connect via phoscon app. Is there any other way to connect a new end-device?
> @Sjonnie2018 : Strange Firmware topic. A0 is the highest number, but 90 has the latest timestamp

I had some trouble getting the device to connect to the Phoscon app. I went to the sensor part in the app and selected Sensors - Other Brand (or something similar) after which I pressed the button on the device for 5 (or 6) seconds to get the device blinking green fast. Then waited about 5 seconds and pressed the button for two seconds. It might have taken two or three tries but eventually the device connected and the app showed the "ready" button. Unfortunately it didn't show up in the app after that though.

About the firmware issue: I wouldn't know what's the best/latest firmware.

@SwoopX and @ebaauw : Can I provide you with more info? Can I test some or other setting that would make the device show up like a sensor? I truly  Hopeappreciate your help so far but I am still stuck here :-| Jope you can help!

SCoe2k, 2020-01-26:
Tks. Still cannot connect. Did exactly as described. No connection possible via phoscon. Even updated firmware to 264A0... Hope the new device request will help.

Is there any way to connect a new zigbee device to decent without phoscon?

SCoe2k, 2020-01-26:
Deventer=deconz

Sjonnie2018, 2020-01-26:
> Tks. Still cannot connect. Did exactly as described. No connection possible via phoscon. Even updated firmware to 264A0... Hope the new device request will help.
> 
> Is there any way to connect a new zigbee device to deconz without phoscon?

You could try the DeCONZ-gui to add devices. The manual that is installed describes how to do that. I haven't tried myself because I am "lucky" enough that the Phoscon app at least discovers the (as yet) unsupported devices.

Sjonnie2018, 2020-02-01:
I have opened a "device support request" #2373 and we are discussing the support for this device over there. To minimize the amount of open issues here I thought it best to close this issue.

closed 2020-02-01 by Sjonnie2018

juul79, 2020-05-12:
Exactly the same problem here.
I'm using ConBee2 + Deconz and Phoscon App on an Intel NUC running Ubuntu shell (without GUI).

Sensors of Xiaomi (door, temp., humidity), Switches of Xiaomi + IKEA and IKEA-lights shown up in Phoscon App... But the HEIMAN Gas- and CO-sensors won't!

The HEIMAN sensors are shown up in Apple HomeKit, using HomeBridge as a bridge between HomeKit and Deconz. So despite they are not visible in the Phoscon App, they seem to be connected.

But activity- and battery-statuses are not updated in HomeKit, so I think they are supported partialy by Deconz/Phoscon at this moment.

**I'm using Version:** 2.05.75  8-3-2020
**Firmware:** 26530700

SwoopX, 2020-05-12:
The device is not visible in Phoscon, as it is closed source and apparently not added. For the value changes, that should have been resolved with the current release. In doubt, redo the binding manually https://discord.com/channels/494922323518947329/704252638727897188/709120485862146161

igeorgie, 2020-05-18:
I can confirm that Heiman devices are **not visible** in Phoscon, but working perfectly within Home Assistant. Phoscon says it has detected a device, but that's it.
Battery level, sensor and switch (for the siren) are detected and can be used in your automations in HA.