dresden-elektronik/deconz-rest-plugin #2324: Xiaomi Aqara pressure sensor reporting less accurate than with Xiaomi gateway

melslenstra, 2020-01-15:
_I initially posted [this issue over at Home Assistant](https://github.com/home-assistant/home-assistant/issues/30769), but got pointed here as the HA deCONZ integration doesn't pick which value to use for sensors._

I'm switching all my sensors from Xiaomi's own gateways (and therefore the xiaomi integration in HA) towards deCONZ. I now have some sensors on Xiaomi's gateway, and some in deCONZ.

The Xiaomi Aqara climate sensor (which reports temp/humidity/pressure) reports its pressure down to one decimal place from Xiaomi, but the deCONZ integration reports it as an integer value (in hPa). I think the deCONZ integration is taking the 0x0000 "Measured value" from deCONZ (which reports an integer in hPa) rather than the more accurate 0x0010 "Scaled value" which in my case reports the extra decimal. 

Some screenshots from the deCONZ GUI:
[image]
[image]
...and finally the result in Home Assistant:
[image]
[image]

According to [this page](https://blog.metservice.com/node/1021), atmospheric pressure is usually between 980 and 1030 hPa at sea level, leaving only 50 discrete integer values of resolution in practice. Although I'm not actually using barometric pressure for anything right now that would really benefit from more resolution, it seems wasteful that the sensor hardware is reporting an extra decimal quite accurately and we're just throwing away that resolution.

I think I can see where in code ([bindings.cpp line 1057](https://github.com/dresden-elektronik/deconz-rest-plugin/blob/10810836905fbcd1e7c9fafcaf386702c9fed41f/bindings.cpp#L1057))the decision is made to use the 0x0000 Measured Value for a pressure sensor. I could probably change that to `0x0010 Scaled Value`, find a way to take `0x0014 Scale` into account, and fix this for my situation. But I'm totally unsure whether any other pressure sensors will always report that, so ideally we would test which properies actually return a value and select the most precise one from that. I'm not familiar enough with the codebase to see if and how we could do such a determination at that point in code however, so any help is much appreciated.

Note: the Phoscon App (deCONZ web UI) also reports the less-precise rounded value:
[image]

SwoopX, 2020-01-15:
Hm, I can follow your thought here. However, I don't see any real benefit. Downside would be, e.g., it would spam my DB storing the values.

Regardless, I had the same thought for power measurement. Precision is also reduced here so everything after the comma is cut off. My personal preference since data is there, but without the mentioned downside as I get changed values anyway every second when a load is applied.

andriej, 2020-01-15:
How about let user decide about the precision?
Data is there to be grabbed, same thing with smoke sensors from Xiaomi/Honeywell...

ebaauw, 2020-01-15:
If memory serves, indeed the REST API plugin uses _Measured Value_, but the code you’re looking for is not in `bindings.cpp`.  That used to setup attribute reporting, which isn’t used for the Xiaomi sensors, since they already report to the coordinator out if the box.

I can confirm that my `lumi.weather` sensors also report _Scaled Value_ in the GUI, so attribute reporting for this attribute works ootb as well.  Exposing this instead thru the REST API is simple, but the REST attribute `state.pressure` currently is an integer reporting the value in hPa.  I have no issues changing that to 0.1 hPa, but that change would be for all pressure sensors.  Also, all REST API clients (Phoscon, but also plugins / bindings for home automation systems) would have to updated to the new scale.

Note: I tried floating point values in REST attributes before, but somehow couldn’t get them to work properly.  I think that’s also why `state.temperature` is an integer in 0.01°C.

SwoopX, 2020-01-17:
> Note: I tried floating point values in REST attributes before, but somehow couldn’t get them to work properly. I think that’s also why state.temperature is an integer in 0.01°C.

I was after thet for the power readings. Also started playing around a bit but obviously no peace of cake...

melslenstra, 2020-01-17:
@ebaauw thanks for clarifying a whole bunch of things so quickly! I see how changing the returned value from the API by a factor of 10 would require changes in all API clients out there as well. I'm nowhere near familiar enough with the deCONZ-REST landscape to have any idea of the scale or feasibility of an operation like that though..

I have used decimal numbers in REST API return objects before without problems. Although decimal number formatting in XML can be tricky (decimal point vs comma etc.), JSON shouldn't give any difficulty due to its [strict number format definition](https://www.json.org/json-en.html). But I'm not sure about any circumstances in this project's context that make this topic more difficult of course.

Arguably, returning a more-or-less standard unit of hPa rather than units of 0.1 hPa might feel a bit neater, but then again if 0.01°C is also used all over the place for temperatures then API client implementors are already going to expect the need to multiply/divide values to get to a commonly used unit/prefix.

In terms of change impact, changing the API to return decimal number hPa values rather than changing it to return integer 0.1 hPa units _might_ save a few client implementations in weak-typed languages from having to change anything (aside from testing). Strong-typed clients would likely have to change their model types to expect a float/decimal type unless they were already using them all along.

melslenstra, 2020-01-18:
@SwoopX 
> Hm, I can follow your thought here. However, I don't see any real benefit. Downside would be, e.g., it would spam my DB storing the values.

The benefit would be in [how barometric pressure is used](https://en.wikipedia.org/wiki/Barometer#Applications) to predict short-term weather changes by looking at rising or falling tendencies. Since these pressure changes are very small (just a few hPa is considered quite significant), only having a resolution of 1 hPa means you spot pressure changes much later or not at all. You can see that my example chart only shows 6 discrete hPa values in a 24-hour period. The weather that day was all over the place around here, from sunshine to stormy and back.

I'll agree that there might be very little use for barometric pressure altogether in a home automation scenario. With the availability of very accurate weather forecast services based on thousands of measurements and advanced algorithms, it makes little sense to try to predict the weather yourself unless you really don't want to depend on *any* internet services. Weather forecasts can be a lot less precise in rural/middle-of-nowhere areas, so it might have its applications there as well.

ebaauw, 2020-01-18:
> I have used decimal numbers in REST API return objects before without problems.

The using isn’t the problem; it’s generating them from the REST API plugin.

jadazul, 2020-02-27:
+1 on this issue. 

I just got my Phoscon gateway and connected the first Aqara sensor, and was dissapointed to find the loss in precision in the pressure. Also it's funny that the temperature and pressure report as integer and the humidity reports with 2 decimals! It'd be great if all of them report with decimals, at least one, even better 2.

Any chance to get this fixed in the next update? 

Thanks in advance!

jadazul, 2020-02-28:
Hi again, if there's no hopes of a fix anytime soon, can you please help with a workaround? 

I'm currently using the deConz integration in Home Assistans, I was thinking in exposing the API using it to get the Scaled Value and divide it by 10... Could you please help me? I'm pretty new to this API (and HA itself).

Thanks in advance! :)

Mimiix, 2020-06-12:
Is this fixed @ebaauw ?

ebaauw, 2020-06-12:
Not by me, and I don’t see any commit referring to this issue.

Mimiix, 2020-06-12:
Let's see if there's any response. @jadazul , let us know!

jadazul, 2020-06-13:
No, unfortunately I haven't heard anything since my last post... :/ I hope someone can help and contribute on this issue to fix it!

Mimiix, 2020-06-13:
@SwoopX or @Smanar  Are you able to look into this ? <3

SwoopX, 2020-06-13:
I'd rather prefer not to do that due to the reasons already outlined by Erik.

ebaauw, 2020-06-13:
We should pick this up in context of APIv2.

Smanar, 2020-06-13:
Lol, sorry same for me, I don't want to broke plugins using a *10 hack.
And change the interger value to a string can be hazardous too.

closed 2020-07-11 by stale[bot]