dresden-elektronik/deconz-rest-plugin #2325: Switch timer by switch

FWillems, 2020-01-16:
Hi,
is it possible to switch a timer by a switch?So, I want to switch a plug on for 10 minutes. After 10 minutes it should automatically switch off. If I press again the switch it should be turned on again for another 10 minutes. 
Thank you.

ebaauw, 2020-01-16:
Best create a rule with a `ddx` condition to turn off the light.

manup, 2020-01-16:
A quick solution is the sensor control editor for that (undocumented feature).

Create a new motion sensor control. Enable the checkmark next to the switch AND also click on the walking man symbol, this actually turns the switch into a motion sensor.

https://phoscon.de/en/app/doc#group-edit-sensors-example-motion-switch-en

FWillems, 2020-01-16:
> A quick solution is the sensor control editor for that (undocumented feature).
>
> Create a new motion sensor control. Enable the checkmark next to the switch AND also click on the walking man symbol, [...]

That sounds pretty easy and should work reliably. Thank you for the good advice.

> Best create a rule with a `ddx` condition to turn off the light.

I don't know how to create a rule. And I also don't know what ddx means. Can you please explain? Perhaps I can also try your solution. Thank you!

ebaauw, 2020-01-17:
You create rules thru the API, see http://dresden-elektronik.github.io/deconz-rest-doc/rules/.  `ddx` is an operator to indicate a given time after last change.  I’m afraid it’s not documented in the deCONZ API, but it is mentioned in the Hue API documentation, https://developers.meethue.com/develop/hue-api/6-rules-api/.

jensflorian, 2020-01-24:
@manup Thanks for the quick solution. This however will not work when you use the switch on the plug itself (or some built-in wall switches). Is there a possibility to trigger the phoscon timer by changing the state of an on/off unit?

closed 2020-05-30 by stale[bot]