dresden-elektronik/deconz-rest-plugin #2328: Not in Network

haasi0000, 2020-01-16:
Hey guys,

I spent a lot of hours without any success.
I use the conbee 2 USB tongle und a raspbee 3 together with hombebridge.
Till yesterday everything worked but now the usb stick can't connect.

The USB Tongle is recognized on the raspberry.

There are my settings. I also uninstall and reinstall the deconz software but also without any success.

[image]

[image]

[image]

Thanks for your help!!!!

Smanar, 2020-01-16:
I don't think it's your response but there is something strange in your config, device type > "router" instead of "coordinator"

manup, 2020-01-16:
Indeed it should be coordinator. Have you tried the load button in the first screenshot?

closed 2020-05-22 by stale[bot]