dresden-elektronik/deconz-rest-plugin #2329: Konke Kit Pro Motion Sensor

dbweb-ch, 2020-01-16:
This sensor was already requested:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1607

But this request was closed without having a working solution.
I've got that sensor and can see it in deconz, but not in the api.
Node Info:
[image]
Cluster Info 0000 Basic:
[image]
Cluster Info 0001 Power Configuration:
[image]
Cluster Info 0500 IAS Zone:
[image]

It gets updated when I move my Hand over (Last Seen gets Updated) so it seems to be working, but how can I get it to the API and Finally to ioBroker?

gijsje, 2020-01-17:
Yes please

Juanezm, 2020-01-20:
👏🏻

realwax, 2020-02-04:
+1 dear Devs, please implement it in deconz and phoscon.

Mimiix, 2020-06-05:
Is this already implemented?

Mimiix, 2020-06-12:
Without a reply on the above, i'll close the issue.

closed 2020-06-12 by Mimiix

Doreinie, 2020-08-17:
H, When will this be working in the Phscon App?
I can see it in Deconz and iobroker. in there surprisingly the battery value appears as 0 %.
If needed for the developper I can provide screenshots, data etc with instruction

SilverWx, 2020-09-15:
I have no battery reported for konke in hassio

Doreinie, 2020-09-15:
Today I see the Konke movement sensor in Phoscon. great and Thanks!

SilverWx, 2020-09-15:
same but no battery