dresden-elektronik/deconz-rest-plugin #2330: deconz doesn't pick up the OTA file

donnib, 2020-01-16:
@manup i have my files in /home/pi/otau

I click on one of the lights in the OTAU list and then Query but it doesn't populate the file so it can update but i can browse to it and ask it to update. What can i do so it picks it up also i can't remember if i have to click each of them and do Query or can it find the files self and start update ?

I run 2.05.72

manup, 2020-01-16:
In case of IKEA OTA files you may have to restart deCONZ once after the files are copied into the otau directory. This is because the files are transformed if needed into a more  standard naming format.

donnib, 2020-01-17:
I did that few times, where should i see the change be made ?

manup, 2020-01-17:
Directly in the otau/ directory, there should be new files witch names like xxxx-yyyy-...zigbee

donnib, 2020-01-17:
[image]
i don't see them, permissions issue or ? is there a log file i can check if there are errors ?

ebaauw, 2020-01-17:
Is the OTAU server enabled?  Check under _Advanced Settings_ in the _Settings_ page in the old web app.

donnib, 2020-01-17:
@ebaauw thx Erik but where is that ? _Advanced Settings_ where ? I looked different places and couldn't find that.

ebaauw, 2020-01-17:
And deCONZ is running under user `pi`?

donnib, 2020-01-17:
I am running the official image from dresden, so i don't know how it's setup.

donnib, 2020-01-20:
more ideas ?

g3wd, 2020-05-12:
I'm having the same issue. 
Deconz is run by user `pi`
otau is enabled in old web interface
deconz 2.05.75
i use a rasbee 1 with a raspi b+ (armv6)

g3wd, 2020-05-15:
Solved it, otau folder was owned by root, changed it to pi:pi `sudo chown pi:pi otau/` restarted deconz and like that the -zigbee files could be created.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix