dresden-elektronik/deconz-rest-plugin #2333: All devices no longer connecting

Mibix, 2020-01-16:
I can see all my lights in the Phoscon web app and in the Deconz GUI but they just aren't connecting no matter what I try.  I've tried unplugging and replugging in the Conbee 2 to my server and it did have my update the firmware once (didn't work before or after though), I've tried closing Deconz and reopening, tried loading a few configurations from the Zigbee configuration page, tried restarting the whole server, and tried flickering on a bunch of lights very close the the Conbee 2. I am on Ubuntu 18.04 LTS using the Deconz GUI I am not sure what else to try. I have had it connect to a few of them once or twice but when I try to control them they always just disconnect immediately.  This has been working for almost a year now it just stopped connecting a day or two ago.   My Conbee 2 is on a usb extension too.
Says Version 2.05.71 and firmware 264A0700 in Phoscon

osos, 2020-01-16:
I have had the same while upgrading from 2.05.69 -> .70.
Ended up downgrading to .67 and re-pairing all.
Others seem to have identified this problem to have been introduced after .67 (maybe .69, maybe .70) several issues in here about it.

@manup, have you ever found the reason for this? Has it been fixed?

Mibix, 2020-01-16:
> I have had the same while upgrading from 2.05.69 -> .70.
> Ended up downgrading to .67 and re-pairing all.
> Others seem to have identified this problem to have been introduced after .67 (maybe .69, [...]

So you didn't downgrade the firmware just the software?  Not exactly sure how to do that, can you let me know how you did it?

osos, 2020-01-17:
I have it running on a Raspberry Pi 3.
Did not do anything with the firmware, I am running the latest (same as you).

Simply uninstalled the .70 version, then reinstalled the .67 stable version. Removed one light bulb and switch at a time, and then re-added into phoscon. My Xiami sensors found themselves back.

Mibix, 2020-01-17:
Think my unit is toast, can't flash firmware in the GUI or via command line and in the GUI it just keeps detecting it then it disappears constantly

meme1337, 2020-01-19:
I'm a recent user of deconz/phoscon and I'm having the same issues. It worked for some days, but now the devices are no longer reconnecting.

osos, 2020-01-20:
@manup, Could this be related to #2349 ?

Is this al about channel changing?

meme1337, 2020-01-20:
> @manup, Could this be related to #2344 ?
> 
> Is this al about channel changing?

I didn’t notice that, but I’ll take care now in case it happens again. Thanks for the heads up. 
In the end I had to reconnect every sensors, it took also surprisingly long to discover the xiaomi window/door ones. 
I need to try rebooting the Linux host of the container: if it gives the error also across reboots I need to find an alternative.

Mibix, 2020-01-20:
I can't even connect to mine anymore :(

Mibix, 2020-01-21:
Well, I took my Conbee II out of my Ubuntu server, installed Deconz on my Windows laptop, opened up deconz, inserted the Conbee II, waited till it prompted me to update the firmware, flashed it in the GUI, got my software all updated on Ubuntu back to the latest version, installed the Conbee II, reset it in Phoscon, repaired all the lights and now it seems to be working again. Epic PITA but its working again ¯\_(ツ)_/¯  I guess the firmware got corrupted or something and it wasn't letting me flash on my Ubuntu server.

Mibix, 2020-01-22:
Still doesn't explain why it just stopped working in the first place since i didn't do any firmware updates before it stopped working but I guess it is fixed now

osos, 2020-02-18:
@manup did put out some response her: https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2423#issuecomment-582514610

> One thing that we working on a while now is to address the software issues which generate the most support requests, like ConBee not connected after restarts and "whole network lost" kind of issues. Hopefully the updates will be out soon and relax the issues.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix