dresden-elektronik/deconz-rest-plugin #2334: Conditional Schedules

marcelmu, 2020-01-16:
I have the lights in my office turning to a warmer light at a given time.
The problem using the time is, that even when I'm not at home the schedule triggers and activates the set up scene.

Is there any way to bind the schedule to a condition? Like only call the scene when the lights are on (another scene is active)?

ebaauw, 2020-01-16:
Two scenario’s
- Set a CLIP flag or status from the schedule, triggering a rule on the flag/status with additional conditions;
- Don’t use a schedule, but a rule with a condition on localtime.  Note that there’s currently a bug in the REST API plugin that might cause the rule not to be triggered, see #2148

marcelmu, 2020-01-16:
Ah great!

I now just created this sensor just containing the flag:
`{
    "config": {
        "on": true,
        "reachable": true
    },
    "etag": "2bfa980fa5d947585c0e9bc8bc6639b3",
    "manufacturername": "Custom",
    "modelid": "T1000",
    "name": "My Switch",
    "state": {
        "flag": true,
        "lastupdated": "2020-01-16T20:08:02"
    },
    "swversion": "1.0",
    "type": "CLIPGenericFlag",
    "uniqueid": "00:00:00:00:00:01-flag"
}`

Then I have this Rule listening on the lastupdated of the sensor with the condition on any light on in the group:
`{
    "actions": [
        {
            "address": "/groups/9/scenes/2/recall",
            "body": {},
            "method": "PUT"
        }
    ],
    "conditions": [
        {
            "address": "/sensors/13/state/lastupdated",
            "operator": "dx"
        },
        {
            "address": "/groups/9/state/any_on",
            "operator": "eq",
            "value": "true"
        }
    ],
    "created": "2020-01-16T20:00:09",
    "etag": "744a363104d0decde2c0139e99dab24b",
    "lasttriggered": "2020-01-16T20:02:12",
    "name": "Switch to evening",
    "owner": "24FC423FB8",
    "periodic": 0,
    "status": "enabled",
    "timestriggered": 1
}`

finally this is my schedule which just updates the sensor:
`{
    "activation": "start",
    "autodelete": true,
    "command": {
        "address": "/api/<API KEY>/sensors/13/state",
        "body": {
            "flag": true
        },
        "method": "PUT"
    },
    "created": "2020-01-16T19:54:05",
    "description": "Trigger sensor flag for warm lights",
    "etag": "6b56c0c61c39328b21be302a0871780c",
    "localtime": "W127/T21:00:00",
    "name": "Abends",
    "status": "enabled",
    "time": "W127/T20:00:00"
}`

Works perfectly! Thank you very much!

Was kind of a hustle getting all the neccesarry data from the preconfigured stuff since the documentation is not flawless.

closed 2020-01-16 by marcelmu