dresden-elektronik/deconz-rest-plugin #2338: Pairing Hue Bulb - Tried it all - not working (Ikea goes fine)

happychriss, 2020-01-18:
Dear,
I am having problems pairing my Hue Bulps with Conbee2, after sending an email to support@dresden-elektronik.de - and not getting any answer - you are my last hope. I hope its OK to post here.

Situation:
I have a fresh Conbee2 ( 2.05.71 12/12/19,Firmware 264a0700) set-up on a RPI3 with Openhabian and managed successfully pairing an Ikea Bulb using http://phoscon.de/
I have also got Deconz working via SSH on the PI (headless set-up), even changed the port of Deconz, so its working together with Openhabian.

All my other lights are Hue that are paired and working to my existing Hue bridge (V1, details below).
I have the Conbee2  and the Hue Bridge direct at the light, trying channels 11 and 25. I know, I have to reset the bulbs to make them visible for Conbee.

**I tried:**
a) Delete Bulb via Hue Bridge - Settings - Info -Delete
b) Use Touchlink via Hue API (http://192.168.1.21/debug/clip.html) and via deconz-api (REST)
c) Via Hue-Bridge - Search for an already connected bulb via serial number

After both, I switched on-off the bulb and started scanning via Poscon-Webpage.

Conbee never sees any bulb.

**Do you have any clue, what I am doing wrong?**

V1 Bridge
"modelid": "BSB001",
"datastoreversion": "59",
"swversion": "01043064",
"apiversion": "1.16.0",

Hue-App: Version 1.1

Hue Bulb
"type": "Dimmable light",
"name": "Hue white light 1",
"modelid": "LWB004",
"manufacturername": "Philips",
"swversion": "5.127.1.26420"

ebaauw, 2020-01-18:
Note that you cannot reset Hue lights using a power-off/power-on sequence; you need to use a Hue dimmer switch or similar to touchlink-reset them.  Alternatively, the Hue bridge will reset the light if you delete it from the bridge (e.g. through the Hue app), but only _while it's reachable_.

Note that Hue lights will only join a ZigBee network, shortly after powered on, even in factory reset state.  Also, they will only respond to touchlink briefly after powered on.

If you're sure the light has been reset and power-cycled properly, and deCONZ still cannot find the light, you want to be looking at (ZigBee) network issues.
- To rule out routing issues, power down all ZigBee routers and keep the light in direct range (< 2 m) of the ConBee to force linking it directly to the ConBee;
- Check for radio interference by WiFi, bluetooth, USB-3, DECT, ...  Disable WiFi and Bluetooth on the computer running deCONZ and disconnect any devices connected to USB-3 ports.  Use a USB extension cable (to a USB-2 port) to keep some distance between the computer and the ConBee.  Don't use a metal enclosure around the light.

happychriss, 2020-01-19:
I tried it all, its working with Ikea via Touchlink and via On-Off. For Hue, none of the ways seems to work. I also use a USB extension cable and the light was direct beside the Conbee. Not sure, what else to try. But thank you very much for your feedback - really appreciated.

closed 2020-05-24 by stale[bot]