dresden-elektronik/deconz-rest-plugin #2339: deCONZ openHAB2 binding with Xioami switch

IsarDomains, 2020-01-18:
Hello,

I've a Xioami switch (`Aqara Xiaomi WXKG11LM Rev.2, Version 20180525`) connected. It's available in the Phoscon app. I'm also able to get the button click/double click/long pressure click values of the switch via the API. In openHAB2 I've installed the binding and configured the stuff like described: https://www.openhab.org/addons/bindings/deconz/. 
However, I'm able to get some property values (channels are available) but I'm not able to react on the button click within an OH rule.

[image]

How I can access the channel of the greyed out "Button Trigger"?
[image]

This rule script does not work for me (channel is never invoked at all):
```
rule "FFAgent Button pressed"
when
 Channel "deconz:switch:conbee2server:xiaomi-ffagent-button:buttonevent" triggered
then
  // do something
end
```

My Things config:
```
Bridge deconz:deconz:conbee2server [ host="192.168.178.32", httpPort="8090", apikey="F2137CAE60" ] {
 Thing switch	xiaomi-ffagent-button	"deCONZ FFAgent Button"	@	"Flur"	[ id="6" ]
}
```

My Items config:
```
Number		FFAgent_Button			"Button [%.0f]"		{ channel="deconz:switch:conbee2server:xiaomi-ffagent-button:button" }
DateTime	FFAgent_LastUpdate		"Letztes Update"			{ channel="deconz:switch:conbee2server:xiaomi-ffagent-button:last_updated" }
Number		FFAgent_BatteryLevel		"Batterieladezustand [%.0f%%]"	{ channel="deconz:switch:conbee2server:xiaomi-ffagent-button:battery_level" }
Switch		FFAgent_BatteryLow		"Batterie FFAgent leer?"		{ channel="deconz:switch:conbee2server:xiaomi-ffagent-button:battery_low" }
Switch		FFAgent_DaylightDark		"Daylight Night or Day"		{ channel="deconz:switch:conbee2server:xiaomi-ffagent-button:dark" }
```

Any thoughts? Thanks.

jensflorian, 2020-01-18:
I think this is not related to deConz. Your rule script is missing the event # - see https://www.openhab.org/addons/bindings/hue/#channels for details. Press the button and check the Openhab log at Port 9001 to see what events are generated.

IsarDomains, 2020-01-19:
Thanks, the issue is that I don't get the button click at all. Even if I enter one of the correct channel numbers (like 1002 or 1004 that I can determine via the deConz Rest-Api) in the rule. 
But I don't see anything "click" related in the logs - even no errors. It's just not there but in contrast I'm able to see and to retrieve the "last update" channel from the switch for example. Maybe I miss something obvious in the configuration for the click event?

closed 2020-05-26 by stale[bot]