dresden-elektronik/deconz-rest-plugin #2340: Trying to get support for new switch

anonomouse, 2020-01-18:
The switch i am trying to add pair is a 6A-RC-ZAB-H0
They come with an Ecosmart bulb, i have several so am willing to send one out to someone smarter than myself to get these implemented. Bulb and remote combo was $9 USD so it seems like a very inexpensive option.

jensflorian, 2020-01-19:
Is this the Zigbee 3.0 Leedarson switch you mentioned ? https://zigbeealliance.org/zigbee_products/leedarson-4keys-remote-controller/

Did you reset the remote before pairing with DeConz? https://fccid.io/2AB2Q6A-RC-ZAB-H0/User-Manual/Users-Manual-3703128 If nothing shows up in the deConz GUI, I would also attempt to touchlink the device with a lamp already included and post the usual screenshots (https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support) here.

anonomouse, 2020-01-20:
Yes that is the switch in question, I have tried pairing / resetting remote and bulb combo many times.
the bulb and remote will function together (out of network).  While pairing the remote the bulb drops from deconz so i can not pull any info for you sorry. Also if i go to add the bulb back into deconz the remote stops functioning.

Smanar, 2020-01-20:
It's something normal, yu can't pair the bulb wit the remote without deconz else they leave the network.
You need to pair the remote, pair the bulb, and put the bulb on the same grup than the remote.
But you need lot of thing before that.
first of them is pair the remote with deconz, the remote can be invisible on phoscon, so you need to take a look in deconz.
https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/Request-Device-Support

anonomouse, 2020-01-23:
I have looked in deconz gui an do not see any unknown devices, i have tried pairing a different remote, same model, to the bulb and i also tried to the usb stick directly, i have not been able to find the device.

thedannymullen, 2020-02-20:
Is this the same remote?  

https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2181

anonomouse, 2020-02-20:
It is, I will look over your thread later,, did you figure it out ?

thedannymullen, 2020-02-20:
I was hoping these could be added.  Apparently they have not yet been.  The other thread appeared to have info but no work was done that I am aware of.

anonomouse, 2020-02-21:
I was reading through a bunch of things last night and was hoping i was
missing something.... Oh well they would be a nice addition since they are
so cheap.
If you find a fix give me a shout, I'll do the same.

thedannymullen, 2020-02-21:
@jensflorian @Smanar the cluster info is in Issue #2181.  Is there more info needed to add this device?

The device request form states something about a debug log, is this required or suggested.

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

raz0rf0x, 2020-06-07:
This issue is not resolved. Remote cannot be paired to deconz. Please reopen and find a solution to the problem.

Smanar, 2020-06-07:
Can you make the same capture than in the other issue to compare the model id for exemple ?

thedannymullen, 2020-07-02:
[image]
[image]

Here are cluster info capture for this remote.  Let me know if I need to provide another one.

Smanar, 2020-07-02:
Ok the switch already included is the "ZBT-Remote-ALL-RGBW", it s the one with color.
The one you have seem easier, and yes I think the cluster list will be usefull.

anonomouse, 2020-07-02:
Please let me know if I can mail anyone one or more of these remotes to
play around with to try to get it to work, I'll even pay shipping 😁

thedannymullen, 2020-07-02:
@Smanar I did not fully understand your response.  Did I provide the correct cluster information?

Smanar, 2020-07-02:
Yep sure, but the cluster list will be usefull too.
It appear when you click on the second circle in the node title.

thedannymullen, 2020-07-03:
[image]
[image]
[image]
[image]
[image]

thedannymullen, 2020-07-03:
@Smanar hopefully I captured the correct ones this time.   Please let me know if you need anything.   See above post and this last screen shot.
[image]

Smanar, 2020-07-03:
lol, so much capture this time ^^
I think I will need some more information later for the battery, but I have enought to start something this WE.
Just 2 questions.
- I m seeing the color control cluster, but why ? this remote have just on/off, light +/ - and a strange button at bottom ?
- I will need you compile the code and give me some return to have button values, do you have an Unix machine and not Hassio ?

thedannymullen, 2020-07-04:
I have access to a Linux machine.  Ubuntu or raspian.  

I can get the info when needed if one of these machines will work.

thedannymullen, 2020-07-04:
[image]

Picture of remote.  The light it was purchased had a color temperature of white, but no colors except white.

Smanar, 2020-07-04:
Ha, it s not exactly the same I have see on the first link.
IDK if the color cluster will be usefull. So I have a code to test (but take care I haven't enought time to compile it on my side, so if you have compilation error, stop the procedure).

You have the procedure here https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only
Just change the step 1 by
`git clone --branch LDS https://github.com/Smanar/deconz-rest-plugin.git`

The battery will not work yet.
The device can be included in the API.
But need to configure button. For that Run deconz in debug mode "--dbg-info=2" will be enought, and press a button , you will see something like that

>08:03:33:302 no button map for: XXXX ep: 0x01 cl: 0x0006 cmd: 0x0A pl[0]: 000

or if you are lucky

>22:11:55:679 button 3001 0 long press

I just need the button you have pressed, if its long press or short press, and the message according to it.

Smanar, 2020-07-07:
Some problem to compile code ?

thedannymullen, 2020-07-09:
Thanks for asking.  I have not had access to the computer I need.  In the process of some home remodels.

I plan to try tomorrow.  Apologize for the delay!

thedannymullen, 2020-07-09:
@Smanar yes I had trouble to compile the code.  

On my raspian-lite install, qmake was not installed so I installed the package: qt4-dev-tools.  Qmake will run now but I get an error in the compile.

```
[11:08:39] mullen@treasurebox:~/deconz-rest-plugin$ qmake && make 
make -f Makefile.Release
make[1]: Entering directory '/home/mullen/deconz-rest-plugin'
/usr/lib/arm-linux-gnueabihf/qt4/bin/uic de_web_widget.ui -o ui_de_web_widget.h
g++ -c -pipe -Wno-attributes -Wno-psabi -Wall -O2 -Wall -W -D_REENTRANT -fPIC -DDECONZ_DLLSPEC=Q_DECL_IMPORT -DARCH_ARM -DARCH_ARMV7 -DGW_SW_VERSION=\"2.05.78\" -DGW_SW_DATE=1593861989 -DGW_API_VERSION=\"1.16.0\" -DGIT_COMMMIT=\"802188c20b12ace030de1bf71384aad0c32fd24d\" -DGW_AUTO_UPDATE_AVR_FW_VERSION=0x260b0500 -DGW_AUTO_UPDATE_R21_FW_VERSION=0x26420700 -DGW_MIN_AVR_FW_VERSION=0x26350500 -DGW_MIN_R21_FW_VERSION=0x26580700 -DGW_MIN_DERFUSB23E0X_FW_VERSION=0x22030300 -DGW_DEFAULT_NAME=\"Phoscon-GW\" -DQT_NO_DEBUG -DQT_PLUGIN -DQT_GUI_LIB -DQT_NETWORK_LIB -DQT_CORE_LIB -DQT_SHARED -I/usr/share/qt4/mkspecs/linux-g++ -I. -I/usr/include/qt4/QtCore -I/usr/include/qt4/QtNetwork -I/usr/include/qt4/QtGui -I/usr/include/qt4 -I../.. -I../../common -Irelease -I. -o release/authorisation.o authorisation.cpp
authorisation.cpp:12:10: fatal error: QMessageAuthenticationCode: No such file or directory
 #include <QMessageAuthenticationCode>
          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
make[1]: *** [Makefile.Release:354: release/authorisation.o] Error 1
make[1]: Leaving directory '/home/mullen/deconz-rest-plugin'
make: *** [Makefile:34: release] Error 2
```

On my ubuntu install qmake was not installed I installed package qt5-qmake, but I still have a qmake not found issue.

Smanar, 2020-07-09:
You haven't forget

>sudo apt install deconz-dev

thedannymullen, 2020-07-09:
Yes I did miss that step.   Looks like it is compiling now!

thedannymullen, 2020-07-09:
Yes I did miss that step.   Looks like it

thedannymullen, 2020-07-09:
When I run the debug command, I see many things coming across but I don't see the "no button map" for this device.  

I am unable to add the remote to the phoscon app under switches and other?  Does it need to be there to capture the button command.

Smanar, 2020-07-09:
Yu have used this line for clone ?
>git clone --branch LDS https://github.com/Smanar/deconz-rest-plugin.git

And yes you can include it usinf "add news switch" but the device can be invisible in phoson, you need to check the API to see it. Using an home automation application that support deconz or a browser with url like
`http://IP:PORT/api/KEY/sensors`

thedannymullen, 2020-07-10:
I used that line for the git clone.  

Then followed the rest of compile instructions you linked to.   So I copied the libxx.so file to the usr directory. 

Then I ran deCONZ with debug option.    I am curious do I need to run the deconz application I just compiled?  I was in the directory I cloned when I started deconz but I don’t know if the one I built started or the one that would be in my path.  Let me confirm and try again in a few minutes.

thedannymullen, 2020-07-10:
@Smanar finally I make some progress.  I made some mistake yesterday, but I have information today.  It may not be the correct information but it is information.  

I am seeing this line for the remote:  ```07:42:00:948 no button handler for: ZBT-CCTSwitch-D0001 ep: 0x01 cl: 0x0008 cmd: 0x00 pl[0]: 0xBF```

I attached more information as a text file.  
[AAA.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4903201/AAA.txt)

I will gather further information as required.

Smanar, 2020-07-10:
Yep I can see too

07:41:17:162 button 1002 Off
07:41:17:326 button 1002 On
07:42:11:970 no button handler for: ZBT-CCTSwitch-D0001 ep: 0x01 cl: 0x0008 cmd: 0x00 pl[0]: 0x7F
07:42:12:185 no button handler for: ZBT-CCTSwitch-D0001 ep: 0x01 cl: 0x0008 cmd: 0x00 pl[0]: 0x40
07:42:12:668 no button handler for: ZBT-CCTSwitch-D0001 ep: 0x01 cl: 0x0008 cmd: 0x00 pl[0]: 0x0D

with same result, bug find, but idk how to correct it ^^, idk for wich one button you had it, you have pressed the same button ?

Your remote have 4 buttons, so can you show me the result for all button (with short and long if the device support it)
For long press, we usualy use the "release" command insted of the "pressed" one.
And tell me the used button, to configure it pls.

thedannymullen, 2020-07-10:
[File2.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4904000/File2.txt)

Here is the result.  I pressed starting at top short press all buttons.  Some took two times to show in the log.  

Then, I long press all buttons starting at top.  One time each.  The only button that did not log data was the last one with the arrow and lines on top of arrow.

Smanar, 2020-07-10:
Ok so first bad news, the 4 th button is not a real zigbee button, it s a programmable button, you can set it yourself.
Zigbee2mqtt are using a hack to manage it, depending on previous used command, and the number of command send.

So I have updated the code, you will have now only (Except for one where you still have "no button handler" but not usefull for me)
>10:25:25:076 button 1002 Some_text_here

So you can confirm me the working one, you will have too useless one, and strange one (with XXXX), it s to find the logic with the 4 th button.

From that I have understand:
- button 3 and 4 will have same command
- the button is always preceded by a "commandMoveToLevelWithOnOff"

thedannymullen, 2020-07-10:
[error.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4905599/error.txt)

I seem to get a compile error this time around.  See attached file.

thedannymullen, 2020-07-10:
[CCCC.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4905685/CCCC.txt)

Here is new log.  I pressed starting at button 1 each button.   I did a short press then long press.

From what I see in the rest API it appears to be working.  Buttons 1-3.  Button 4 shows:

```7:13:21:266 verify 0xccccccfffee3ab17 is child node after 785748 s
17:13:21:266 Force binding of attribute reporting for sensor ZBT-CCTSwitch-D0001 13
17:13:21:266 no button handler for: ZBT-CCTSwitch-D0001 ep: 0x01 cl: 0x0008 cmd: 0x04 pl[0]: 0xFE
17:13:21:271 verify neighbor status: APP_SUCCESS (0x00) 
17:13:21:276 APS-DATA.indication from child 0x0968
17:13:21:277 button 3002 Temperature short
```

SwoopX, 2020-07-10:
@thedannymullen @Smanar forgot a comma in `sensor.cpp`, line 774 after `0xFE`. You may want to add it yourself and try again.

thedannymullen, 2020-07-10:
**Update from Rest API**
More Data:
Button one has short press only code 1002.  Long hold, then release shows 1002.  
Button 2 has 2001,2002,2003.  I did notice it requires one short press for the long press 2001 code to show after pressing another button.  Code 2003 will show which is long release, but not the long press. 
Button 3 has 3002 only.

Button 4 shows 3002 code in rest API.

thedannymullen, 2020-07-10:
[DDDD.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4905778/DDDD.txt)

This log is after the recompile with the comma on line 774.  I pressed every button short press one time.  Then press button 2 long press and button 4 short press again.

Smanar, 2020-07-11:
I don't see long press for the button 1, are you sure having one, it s a button for on/off, I don't see why you need a long press on it ?

I don't understand 
>it requires one short press for the long press 2001 code to show after pressing another button

On log you have
>17:36:47:226 button 2001 Dim long press
>17:36:47:966 button 2003 Dim long release

So it s not good ?

For button 3, it s strange, what have you when doing long press ? On your previous log, you have some more command

For the button 4, I think I have an easier solution
>17:36:50:538 button 2001 XXX

It seem you have it only when using button 4 ?
And have after

>17:36:50:544 button 3002 Temperature short

thedannymullen, 2020-07-11:
Button 1 does not need a long press.  I was commenting my findings.  It works correctly.  

Button 2 has a strange behavior for the long press the 2001 code long press did not seem to show unless I do a short press of button 2 first.   I am not worried about this currently.   It was something I noticed.  

I can try button 3 long press again and log behavior.  The logs get very big it is possible I missed something in my separation.  

Button 4 showed the 2001 xxx and the 3002 command.

Smanar, 2020-07-11:
So if you make directly a long press on the button 2, you have no event "long push" and "long push release" ?
Idk how work the initial remote, you have already used it with a classic bulb ?
Nothing special in logs ?

Ok so for button 4, I will use the XXXX command and ignore the 3002 just after.

For the button 3, I don't need full log, just look at the log and tell me if you see a missing command. But the problem can be from me too, there is some lack for cluster 0x0300 in code. If you realy see nothing just tell me.

thedannymullen, 2020-07-11:
> So if you make directly a long press on the button 2, you have no event "long push" and "long push release" ?
Idk how work the initial remote, you have already used it with a classic bulb ?

Seems to work now.  Sometimes with debug logging on things can be slow.  

Button 4 logic sounds good to me.  

I will look for the button3 in log shortly.

Smanar, 2020-07-13:
Do you want to make a new try with the button3 or I try first a code for the button 4 ?

guedeaux, 2020-07-13:
I have this switch. I see it in VNC but not in Phoscon (which I think is expected); I am running the Deconz Add-on in HomeAssistant on a Linux machine. Anything I can do to help?

thedannymullen, 2020-07-13:
> Do you want to make a new try with the button3 or I try first a code for the button 4 ?

I am willing to try button 4.  I will try a long press on button 3 right now also.  

@smanar, I will be unavailable tomorrow to try anything.  But free again on Wednesday.

thedannymullen, 2020-07-13:
@Smanar If I understand the code in sensor.cpp I see what looks like this command, but it is not picking up the command as a long press.  But this was a long press of button 3.  

Button3 long press capture:  ```
8485:16:10:56:641 Force binding of attribute reporting for sensor ZBT-CCTSwitch-D0001 13
18569:16:10:56:641 no button handler for: ZBT-CCTSwitch-D0001 ep: 0x01 cl: 0x0300 cmd: 0x4B pl[0]: 0x03
18667-16:10:56:650 verify neighbor status: APP_SUCCESS (0x00) 
18724-16:10:57:132 MAC Poll 0x02 0x0968
18758-16:10:57:137 poll node d0:cf:5e:ff:fe:23:ee:50-01
18808-16:10:57:137 Poll light node Color1
18844-16:10:57:186 Poll APS request to 0xD0CF5EFFFE23EE50 cluster: 0x0006 dropped, values are fresh enough
18945-16:10:57:224 giveup binding srcAddr: B57FFFE489D72
18996-16:10:57:276 Bind response success for 0xd0cf5efffe7cb52f ep: 0x01 cluster: 0x0008
19079-16:10:57:276 skip configure report for cluster: 0x0008 attr: 0x0000 of node 0xD0CF5EFFFE7CB52F (wait reading or unsupported)
19204-16:10:57:285 Websocket 192.168.107.20:55528 send message: {"attr":{"lastannounced":null,"lastseen":"2020-07-13T21:10:57Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI control outlet","name":"Christines Lamp","swversion":"2.0.022","type":"On/Off plug-in unit","uniqueid":"d0:cf:5e:ff:fe:7c:b5:2f-01"},"e":"changed","id":"7","r":"lights","t":"event","uniqueid":"d0:cf:5e:ff:fe:7c:b5:2f-01"} (ret = 340)
19615-16:10:57:502 Poll APS request to 0xD0CF5EFFFE23EE50 cluster: 0x0008 dropped, values are fresh enough
--
23148-16:10:58:436 DB saved in 17 ms
23179-16:10:58:436 don't close database yet, keep open for 900 seconds
23244-16:10:58:637 Poll APS request to 0xD0CF5EFFFE22E852 cluster: 0x0008 dropped, values are fresh enough
23345-16:10:58:732 Poll APS request to 0xD0CF5EFFFE22E852 cluster: 0x0300 dropped, values are fresh enough
23446-16:10:59:278 Bind response success for 0xd0cf5efffe7cb52f ep: 0x01 cluster: 0x0000
23529-16:10:59:284 Websocket 192.168.107.20:55528 send message: {"attr":{"lastannounced":null,"lastseen":"2020-07-13T21:10:59Z","manufacturername":"IKEA of Sweden","modelid":"TRADFRI control outlet","name":"Christines Lamp","swversion":"2.0.022","type":"On/Off plug-in unit","uniqueid":"d0:cf:5e:ff:fe:7c:b5:2f-01"},"e":"changed","id":"7","r":"lights","t":"event","uniqueid":"d0:cf:5e:ff:fe:7c:b5:2f-01"} (ret = 340)
23940-16:10:59:399 poll node 00:0b:57:ff:fe:4c:82:c0-01
23990-16:10:59:400 Poll light node Back1
24025-16:10:59:453 Poll APS request to 0x000B57FFFE4C82C0 cluster: 0x0006 dropped, values are fresh enough
24126-16:10:59:645 APS-DATA.indication from child 0x0968
24177:16:10:59:646 Force binding of attribute reporting for sensor ZBT-CCTSwitch-D0001 13
24261:16:10:59:646 no button handler for: ZBT-CCTSwitch-D0001 ep: 0x01 cl: 0x0300 cmd: 0x4B pl[0]: 0x00
24359-16:10:59:990 Websocket 192.168.107.20:55528 send message: {"attr":{"lastannounced":"2020-07-12T14:17:02Z","lastseen":"2020-07-13T21:10:59Z","manufacturername":"Philips","modelid":"LTW013","name":"Kitchen Track 2","swversion":"1.46.13_r26312","type":"Color temperature light","uniqueid":"00:17:88:01:04:e9:81:ab-0b"},"e":"changed","id":"10","r":"lights","t":"event","uniqueid":"00:17:88:01:04:e9:81:ab-0b"} (ret = 347)
24777-16:11:00:139 MAC Poll 0x02 0x0968
24811-16:11:00:326 Bind response success ```

Smanar, 2020-07-14:
Ok so I have modified the code (not tested yet, so if you have compliation error, stop here)

For button 4, I m memorizing the last command and ignore the second one if I have the first one.
I have added some more debug line for button 3, the deconz code make some "magic" with values, so hard to guess the good one, so I have added a debug line that print the value I need.

You will have in log
`LDS debug 2:  XXX`

thedannymullen, 2020-07-14:
@Smanar thanks!  I can try tomorrow for sure.  I have very limited computer access today.

thedannymullen, 2020-07-14:
[AAA.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4918935/AAA.txt)

I did have a little time to try the compile.   Looks like there was an error.

Smanar, 2020-07-14:
My bad, corrected, tested, it compile now.

thedannymullen, 2020-07-15:
Here is my log:

I could not figure out exactly how to parse it out.  I saw some 4001 codes in the log when I pressed button 4. 
[dbuginfo.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4928580/dbuginfo.txt)

Smanar, 2020-07-16:
So the button 4 is working  ?

The debug was added for button 3, to find long press, and I don't see button 3 event ?

thedannymullen, 2020-07-16:
Yes button 4 gets a code of 4001 in the rest api.  It appears to be working.  

Button 3 does not appear to work in the rest api.  I tried to capture 4 button 3 events in the log.  I short press 2 times then long press two times.

[dbuginfo.txt](https://github.com/dresden-elektronik/deconz-rest-plugin/files/4932584/dbuginfo.txt)

Smanar, 2020-07-16:
The button 4 not have long or short, sorry. In fact by construction it's not a real button, It just a scenario memory.

So the button 3 is broken ? I don't see event at all, need to compare with working version.  But I m looking the code and just this part
```
            //to remove
            if (ind.clusterId() == COLOR_CLUSTER_ID &&
                     (zclFrame.commandId() == 0x4b ))
            {
                DBG_Printf(DBG_INFO, "LDS debug 1");
            }
```

Can't broke the code, you you need have in log "LDS debug 1" somewhere. ATM on your log you have only log when zclFrame.commandId() == 0x0A.

Perhaps your device have loose some bind ?

thedannymullen, 2020-07-16:
Yes button 4 I short press am doing long press both show 4001 in the rest api.   Maybe it is like button one with no long press?

I will try button 3 again and log.  I don’t know off  saved an old version of the code to test I will look.

Smanar, 2020-07-17:
If you want to disable new code, in de_web_plugin.cpp
line 3805
```
            else if (ind.clusterId() == COLOR_CLUSTER_ID &&
                     zclFrame.commandId() == 0x0a && zclFrame.payload().size() >= 2 && // Move to Color Temperature
                     sensor->modelId().startsWith(QLatin1String("ZBT-abshgdhdhgdghdgdg")))
            {
```

But if you just try to include it again, it will remake all binds.

thedannymullen, 2020-07-18:
@Smanar I commented line 3805 - 3819 doing this will make button 3 work for short press, but button 4 appears to not show in the rest api as working.

[40 more replies]

closed 2020-08-13 by manup