dresden-elektronik/deconz-rest-plugin #2342: Android Apps that work with Deconz or Conbee or Phoscon ?

Ltek, 2020-01-19:
Anyone know of Android apps that work with our devices?

If you know of android apps that work - please pose HOW you connect them to Deconz. 

I've tried 3 different top Android Hue apps and they never find Deconz even if I put in the IP & port.

thx!

FYI - Hue Essentials advertises Deconz support and it works... but has 2 major problems that is making me look for something else...
1. UI is VERY slow to update light status (colors, dim level, etc). 5-10 seconds or longer. Seems like it doesnt force an immediate update of all lights in a given scene, or even of the individual light itself when you set that light. Looks like it waits for a polling period or something else. -- really annoying!
2. not efficient for frequent use. layout and navigation is tedious and has no real customization (cant create a home screen with scene buttons, or such. ... but after waiting and making suggestions for a year, its not gotten any better.

Smanar, 2020-01-19:
You can use directly phoscon, but I think you will not like the UI.
But you can use a complete home automation application, I m sure in this situation you can find exactly the UI you want.

vakilando, 2020-05-20:
hue essentials (https://www.hueessentials.com/) works with deconz.
But I never tried it...

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix