dresden-elektronik/deconz-rest-plugin #2344: Cannot get stable connections, cannot change channel.

itsjakey, 2020-01-19:
Hello,

I have started using DeConz-Rest-Plugin for around 3 weeks now.
I have a RaspBee running on a RaspberryPi 4, using DietPi.

Now eventually with a struggle, I got everything working well. Then randomly lights would not respond, a reboot would fix them. Then the whole system lost connection to 80% of lights. Reboots etc would not being it back.

So I started again, re-paired everything, and the same scenario happened again.

What I have noticed is that things work most happily when it uses Channel 11.
So I reset everything today, including the Phoscon Gateway.

It initially booted on channel 11, then switched to channel 15 without me doing anything.

I tried to pair some lights, nothing found.
I tried to pair a switch, nothing found.

So I tried to switch back to channel 11, and although I reports success, when you go back into the advanced options, its back on channel 11. So I thought I'd try Channel 20 or 25. They all do the same (switching back to 15).

In the log files I see the following;
│ Jan 19 20:21:58 DietPi deCONZ[1084]: 20:21:56:504 channel 15 does not match  │
│ channel mask 0x02000000 (TODO)                                               │
│ Jan 19 20:21:58 DietPi deCONZ[1084]: 20:21:56:504 channel is 15 but should   │
│ be 25, start channel change                                                  │
│ Jan 19 20:21:58 DietPi deCONZ[1084]: 20:21:56:504 Skip automatic channel     │
│ change, TODO warn user   

Can anyone give me some help around what is happening?

At the moment I'm stuck, I just want a stable home system. 

I am using Ikea lights (5 GU10's, and 1 E27, and 1 of the smaller E. I also have ikea smart plugs and a repeater.) Although it was unreliable before the smart plugs and repeater was added.

Larsn1, 2020-01-20:
I had the same issue yesterday. I cannot pair Hue lights. My gateway is on channel 25 and the hue lights switch to channel 11 as soon as they are back in their lamps.

Larsn1, 2020-01-20:
One more thing I noticed is that my network has switched the network Id without a reason. Had to repair the system for the whole weekend ... 2.5.72

itsjakey, 2020-01-20:
I am on 2.5.71, I take it you are running Docker?
I am running it on the native install of dietpi.

How do you work out what channel the lamps are on?

Even when I switch the configuration being used in the advanced settings to an older one, the lights still do not respond.

At the moment can’t talk to the lights at all.

Larsn1, 2020-01-20:
I scan in the old web interface with the touchlink reset tool and see that the lamps are in the network but on a different channel. I don’t know how to force them to use the coordinators channel

itsjakey, 2020-01-21:
Okay, so now I've re-imaged everything again, this time using Docker (marthoc/deconz).
Initially, no better off really, wasn't finding lights.

So hard powered it down, disconnected the raspbee and placed it on a bit of a flylead to take it 15cm from the Pi. Then booted everything up and it started finding lights.
Docker image stayed on channel 11 so far this time.

Only thing to note, is that my Pi4 has a case on it, large metal heatsink type case. However, the Raspbee is on a gpio extender and sits outside and around 3cm above this heatsink case.
Bluetooth and Wifi disabled on the Pi.
Disablement of Wifi in the house. No affect.

So either the Raspbee required a hard power off to start working in Docker, or it is related to being close to a raspberry pi 4.

I will see if it lasts more than a week now and keep this thread updated.

closed 2020-05-29 by stale[bot]