dresden-elektronik/deconz-rest-plugin #2345: Unable to change the state of a motion sensor via API

ronaldheft, 2020-01-19:
I'm attempting to change the state of a motion sensor using:
`PUT /api/<apikey>/sensors/<id>/state`

When I make the following request, I receive a blank response and the state does not change:

Request:
```
PUT /api/<apikey>/sensors/<id>/state HTTP/1.1
Content-Type: application/json

{
    "presence": false
}
```

Response:
```
[
    {}
]
```

Notice the weird response. I don't receive an outright error, just a blank response. I am able to update attributes of the device using the `PUT /api/<apikey>/sensors/<id>` endpoint, so the API is working fine; I just can't update the state.

**Device in Question:** Philips Hue Motion Sensor
**Device Firmware:** 6.1.1.27575
**deconz Version:** 2.05.72
**Gateway:** Conbee II
**Gateway Firmware:** 264A0700

ebaauw, 2020-01-19:
You can only update state attributes of CLIP sensors; for ZigBee sensors they’re read-only,  The error reporting by the API needs to be improved.

ronaldheft, 2020-01-19:
Thanks for the help! That makes sense. Happy to close this or we can leave it open if there is the plan to improve the error message.

closed 2020-05-26 by stale[bot]