dresden-elektronik/deconz-rest-plugin #2346: Activate power on behavior

broliyoung, 2020-01-19:
I'd like to manage the power-on behavior.

wvuyk, 2020-01-19:
which power on behavior? Of deCONZ? Of Conbee? Of Raspberry, of a bulb? Or a sensor? Or your house?

Pirlobeck, 2020-01-20:
I think, broliyoung means the behavior of the bulbs ,when they get power again, after powerloss.

broliyoung, 2020-01-20:
Thanks @Pirlobeck for explain of my question. Sorry @wvuyk for missing precious information for your undarstanding. Since if I search "power on behavior" on Google, first 10 pages talk about the bulbs, I though it was a special feature that is not confusingly with other things like my home or deConz or Conbee that actually already have automatic start.

So, what I need is to manage the feature on my philips HUE and maybe on IKEA BULBS to stay OFF   when they get power again, after powerloss.

I sow there is another issue opened with same topic but it is marked Stale whithout solution

danielkihlgren, 2020-02-03:
If possible it would be nice to toggle the light from previous state when it had power.

mtornblad, 2020-03-01:
+1 on this is the thinking is to add the following setting into the webgui! Same setting should work on Hue with newer firmware also as i understand it! Writing "last scene" to Ikea fails, while on and off seems to work now after their latest firmware! (Lot of people don't like behaviour change that they power on as Off instead of On as last firmware (Ikea bug?))

[image]

mgrn0, 2020-05-02:
I tried to set the attribute to 'Previous' through the deconz UI shown above but I get a 'writing failed' error; using latest .75 version. 

The bulbs are running lates Hue firmware and did proberly work using the 'previous' setting and a Hue switch (tested this to rule out problems regarding the bulbs).

@mtornblad did writing this setting work for you?

mgrn0, 2020-05-02:
ah, just found https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1055#issuecomment-559993258 - at least there is a way to do it, even if that is not really good UX and should be supported by the web interface hopefully at some point in time.

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

axlthorell, 2020-09-26:
Hi, just thinking would it be possible to have On/Off behavior as opposite of previous state?
It would make the (mechanical) wall switches useful for IKEA lights, if a spring is used with the switch. 
Push to toggle the power quickly and turn the light on/off.

Smanar, 2020-09-26:
From my memory ikea device don't support this feature, you see it on deconz and use it, but the setting is loose when the device is power off, making it useless.

mtornblad, 2020-09-28:
> From my memory ikea device don't support this feature, you see it on deconz and can use it, but the setting is loose when the device is power off, making it useless.

The later IKEA-firmwares have the option to set and save power-on behavior. I think it was possible to set as on, off, or last-scene!

axlthorell, 2020-09-28:
> The later IKEA-firmwares have the option to set and save power-on behavior. I think it was possible to set as on, off, or last-scene!

The feature that I'm looking for is to set the behavior to opposite of last-scene. This would enable the use of ordinary wall switches to both turn on and off the lights and not leave them without power.
Is this supported in by the Zigbee standard?

Mimiix, 2020-09-28:
@axlthorell Please open a new user question issue instead of necroposting this one :)

axlthorell, 2020-09-28:
@Mimiix to be continued in #3330 :)

oudoulj, 2023-12-03:
> I tried to set the attribute to 'Previous' through the deconz UI shown above but I get a 'writing failed' error; using latest .75 version.
> 
> [image]
> 
> I used a regular hue bulb (just lightness and temperature adjustable).
> 
> The bulbs are running lates Hue firmware and did proberly work using the 'previous' setting and a Hue switch (tested this to rule out problems regarding the bulbs).
> 
> @mtornblad did writing this setting work for you?

I have the same problem over here with a LIDL Livarno Lux bulb : I get "writing failed" when trying to change the value of `StartUp OnOff`.
My goal is to have the bulbs `off` after a power failure instead of always `on`.
Should I modify `StartUp OnOff` or `Power on state` (that one is currently `off` so I don't think that's the one I need after a power failure)
How do I go by with that "writing failed" though? 😿 
Thank you!

[image]

Smanar, 2023-12-04:
And if you try  the "read" request, the attribute is disabled (grayed) or stay enabled ?

oudoulj, 2023-12-04:
> And if you try the "read" request, the attribute is disabled (grayed) or stay enabled ?

I double-clicked the attribute, tried to change it from on to off, got "writing failed" then clicked on Read and got something along the lines "unsupported attribute" and now the line is greyed out.
I already got that for all attributes earlier and had to quit Deconz and come back to make them available again but after a couple trial and errors.
I noticed that those attributes do not refresh according to the device I select on the right on my mesh network.
Either I'm not doing things right or the interface is a little flaky.

[image]

Smanar, 2023-12-05:
Generaly grayed mean not supported, so can explain your issue.

>I already got that for all attributes earlier and had to quit Deconz and come back to make them available again but after a couple trial and errors.

Yeah to test, better to select only 1 attribute.