dresden-elektronik/deconz-rest-plugin #2347: Issue with decina on home assistent

rorshark01, 2020-01-20:
Hello everybody,

I'am new of the forum and i really Need tour help of you can. I try ti setup deconz on homeassisten on rasperrypie. Everything work by installing app deconz and login. Now when i log in in the deconz app directory with home assistente link the io showed Is corret: my ip. After insertion of password i enter in the app bit the ip showed in the top fight of the screen Is changed and no interaction with my xiaomi sensor. Id otherwise i try to log in with the originale homepage of deconz every thing work fine and the ip Is corret.
Do giù can help me?
Thank you

chbndrhnns, 2020-01-20:
@rorshark01 Would you say that this is your issue? https://github.com/home-assistant/hassio-addons/issues/947

Do you try to access it via the home assistant app or directly with IP and port?

rorshark01, 2020-01-20:
Hello
I don't know how but Just refresh now It work
Thank you
You can close the topic

closed 2020-05-26 by stale[bot]