dresden-elektronik/deconz-rest-plugin #2348: ConBee II (coordinator) Not able to associate with other devices

kentwinther, 2020-01-20:
Hi all,
I have an issue with my ConBee II that it does not associate with any devices anymore...
It has been working, but I did a reset to in order re-configure my whole setup.

My "simple" setup is now:
* DeCONZ/Phoscon: 2.05.71
* ConBee II firmware: 264A0700
* ConBee II as "Coordinator"
* and for now just one bulb (Philips) which has been reset.

I have reset the ConBee II via Phoscon web interface a couple of times, but this does not help.

In Wireshark (sniffing ZigBee) I do see the following:
1. the bulb sends: Beacons
2. the bulb sends: Association Request
3. ConBee II sends: Association Response with 
- Short Address = 0xFFFF
- Association Status = 0x01 (PAN Full / PAN at capacity)!!!
So it seems like that my ConBee II is FULL - not able to associate any more.
But how I do I reset the ConBee so that the "stored" PANs are also removed?
Or what to do? Any ideas?

Thanks in advance.
Kent.

kentwinther, 2020-01-22:
Hi again,
Is there anyone who can help me with my ConBee II issue as described above?
Any advice will be appreciated.
Thanks

Smanar, 2020-01-22:
You have succed with only 1 bulb, and no more device ?
Can you show the configuration in deconz ?
Have you try to run deconz in debug mode ?
Have you tried to restore a previous configuration (cf https://github.com/dresden-elektronik/deconz-rest-plugin/releases?after=V2_05_62 ) ?

kentwinther, 2020-01-22:
Hi
@Smanar 

> You have succed with only 1 bulb, and no more device ?

My ConBee II has been working and associated with different devices. But I wanted re-configure my whole setup so I did a reset of my ConBee II via Phoscon. After that I have not been able to associated any devices to my ConBee II - not even devices that has been associated with my ConBee II before the reset.

> Can you show the configuration in deconz ? 

Yes:
[image]

> Have you try to run deconz in debug mode ?
Yes (deCONZ.exe --dbg-info=1 --dbg-aps=1 --dbg-zcl=1 --dbg-zdp=1), but it does not give any hints on the issue - at least from my point of view :) 
Log:
`00000001	13:16:10	[6108] libpng warning: iCCP: known incorrect sRGB profile	
00000002	13:16:10	[6108] 13:16:10:300 HTTP Server listen on address 0.0.0.0, port: 80, root: C:/Users/Bruger1/AppData/Local/deCONZ/bin/plugins/de_web/	
00000003	13:16:10	[6108] 13:16:10:317 CTRL. 3.24.0	
00000004	13:16:10	[6108] 13:16:10:408 ZCLDB init file C:/Users/Bruger1/AppData/Local/dresden-elektronik/deCONZ/zcldb.txt	
00000005	13:16:10	[6108] 13:16:10:427 gw run mode: normal	
00000006	13:16:10	[6108] 13:16:10:427 GW sd-card image version file does not exist: C:/Users/Bruger1/AppData/Local/dresden-elektronik/deCONZ/gw-version	
00000007	13:16:10	[6108] 13:16:10:428 DB sqlite version 3.24.0	
00000008	13:16:10	[6108] 13:16:10:428 DB PRAGMA page_count: 30	
00000009	13:16:10	[6108] 13:16:10:429 DB PRAGMA page_size: 4096	
00000010	13:16:10	[6108] 13:16:10:429 DB PRAGMA freelist_count: 0	
00000011	13:16:10	[6108] 13:16:10:429 DB file size 122880 bytes, free pages 0	
00000012	13:16:10	[6108] 13:16:10:429 DB PRAGMA user_version: 6	
00000013	13:16:10	[6108] 13:16:10:429 DB cleanup	
00000014	13:16:10	[6108] 13:16:10:429 DB create temporary views	
00000015	13:16:10	[6108] 13:16:10:433 don't close database yet, keep open for 900 seconds	
00000016	13:16:10	[6108] 13:16:10:434 started websocket server at port 443	
00000017	13:16:10	[6108] 13:16:10:448 found node plugin: de_rest_plugin.dll - REST API Plugin	
00000018	13:16:10	[6108] 13:16:10:450 found node plugin: de_signal_plugin.dll - Signal Monitor Plugin	
00000019	13:16:10	[6108] 13:16:10:461 found node plugin: std_otau_plugin.dll - STD OTAU Plugin	
00000020	13:16:10	[6108] 13:16:10:462 watch: C:/Users/Bruger1/AppData/Local/deCONZ/bin/plugins/de_web	
00000021	13:16:10	[6108] 13:16:10:462 watch: C:/Users/Bruger1/AppData/Local/deCONZ/bin/plugins/de_web-src	
00000022	13:16:10	[6108] 13:16:10:474 found node plugin: templ_plugin.dll - Templ Plugin	
00000023	13:16:10	[6108] 13:16:10:714 New websocket 192.168.0.60:63342 (state: 3) 	
00000024	13:16:10	[6108] 13:16:10:790 Device firmware version 0x264A0700	
00000025	13:16:10	[6108] 13:16:10:805 unlocked max nodes: 200	
00000026	13:16:10	[6108] 13:16:10:898 Device protocol version: 0x010B	
00000027	13:16:10	[6108] 13:16:10:908 new node - ext: 0x00212effff050445, nwk: 0x0000	
00000028	13:16:10	[6108] 13:16:10:972 ZDP node descriptor request to 0x00212EFFFF050445	
00000029	13:16:10	[6108] 13:16:10:972 APS-DATA.request id: 4, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0002, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x04	
00000030	13:16:10	[6108] 13:16:10:972 ZDP send request id: 0x03 to 0x00212effff050445	
00000031	13:16:11	[6108] 13:16:11:006 Current channel 13	
00000032	13:16:11	[6108] 13:16:11:023 CTRL ANT_CTRL 0x03	
00000033	13:16:11	[6108] 13:16:11:051 Device protocol version: 0x010B	
00000034	13:16:11	[6108] 13:16:11:101 Current channel 13	
00000035	13:16:11	[6108] 13:16:11:119 CTRL ANT_CTRL 0x03	
00000036	13:16:11	[6108] 13:16:11:164 APS-DATA.confirm id: 4, status: 0x00 SUCCESS	
00000037	13:16:11	[6108] 13:16:11:164 APS-DATA.confirm request id: 4 -> confirmed, timeout 1579695370	
00000038	13:16:11	[6108] 13:16:11:168 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8002, lqi: 241, rssi: 19	
00000039	13:16:11	[6108] 13:16:11:168 APS-DATA.indication request id: 4 -> finished	
00000040	13:16:11	[6108] 13:16:11:168 APS-DATA.request id: 4 erase from queue	
00000041	13:16:11	[6108] 13:16:11:168 ZDP status = 0x00 -> SUCCESS	
00000042	13:16:11	[6108] 13:16:11:168 ZDP Node_Descriptor_rsp 0x00212EFFFF050445 - 0x0000	
00000043	13:16:11	[6108] 13:16:11:347 ZDP active ep request to 0x00212effff050445	
00000044	13:16:11	[6108] 13:16:11:347 APS-DATA.request id: 6, addrmode: 0x02, addr: 0x0000, profile: 0x0000, cluster: 0x0005, ep: 0x00 -> 0x00 queue: 0 len: 3 tx.options 0x00	
00000045	13:16:11	[6108] 13:16:11:347 ZDP send request id: 0x07 to 0x00212effff050445	
00000046	13:16:11	[6108] 13:16:11:360 APS-DATA.confirm id: 6, status: 0x00 SUCCESS	
00000047	13:16:11	[6108] 13:16:11:360 APS-DATA.confirm request id: 6 -> confirmed, timeout 1579695371	
00000048	13:16:11	[6108] 13:16:11:364 APS-DATA.indication srcAddr: 0x0000, srcEp: 0x00 dstAddrMode: 2, profile: 0x0000, cluster: 0x8005, lqi: 241, rssi: 19	
00000049	13:16:11	[6108] 13:16:11:364 APS-DATA.indication request id: 6 -> finished	
00000050	13:16:11	[6108] 13:16:11:364 APS-DATA.request id: 6 erase from queue	
00000051	13:16:11	[6108] 13:16:11:364 ZDP status = 0x00 -> SUCCESS	
00000052	13:16:11	[6108] 13:16:11:364 ZDP active ep response for 0x00212effff050445	
00000053	13:16:11	[6108] 13:16:11:364  ep: 0x01	
00000054	13:16:11	[6108] 13:16:11:364  ep: 0xF2	
[... cut ...]
And then APS-DATA.request  and  APS-DATA.indication request continues like this forever...

> Have you tried to restore a previous configuration?

Yes, but still not working. There is only one entry as of now:
[image]

As mentioned about I do see the following in Wireshark when sniffing:
The setup is ConBee II and one bulb that has been reset, and then changed "PermitJoin" to 255 via deCONZ.

[image]

It looks like the ConBee II rejects devices because it "thinks" it full (no room for new PANs)...

Any other suggestions?

kentwinther, 2020-01-22:
Wrong picture above of Wireshark... The right one is here:

[image]

closed 2020-05-30 by stale[bot]

elbrini, 2020-10-04:
I have the same issue here.
Coordinator answers with an unsuccessful association response. The association status is 0x02 which means the PAN is full.
I tried also to reset the ConBee2 gateway but without success.
Any suggestions?

Mimiix, 2020-10-06:
@elbrini No issue is ever the same. Can you please open a own issue?