dresden-elektronik/deconz-rest-plugin #2349: Network Id is changing randomly

Larsn1, 2020-01-20:


Larsn1, 2020-01-21:
Tried to add the log yesterday evening just to find out that the logging was disabled. Jumped back to 2.5.69 because of fyrtur blinds and added one room again. Let’s see what happens today.

Larsn1, 2020-01-21:
Network I’d is the same as yesterday. Everything is still reachable and running well. Seems like 2.5.69 is safe and breaking changes have been made later.

Larsn1, 2020-01-23:
Two days and everything is running well. As I havent seen a commit that could influence that it probably is a misconfiguration on the user side. Maybe you can find out what could have caused this. Meanwhile I stay back from newest versions.

easybeat, 2020-01-28:
Hi,
I think I did face the same problem after I did update to 2.5.72 on Rasbee and didn't understand at all what it caused. But reading your post I understand.
I am back on 2.5.64, rejoined every device and it is working fine since 2 weeks now!
Thanks

Larsn1, 2020-02-03:
Everything is still working. Unfortunately I don’t expect the developers to answer here because we don’t have the log of the moment where the device is changing the network id. I hope it was helpful to narrow down the version that might could be affected

closed 2020-06-09 by stale[bot]