dresden-elektronik/deconz-rest-plugin #2350: Develco?

eddiex666, 2020-01-20:
Hi,

sorry for this question here, i was'nt able to find a other forum from deconz.

I would like to get all my devices from deconz into Domoticz or some other smart home software.
I have connected the plugins, but i can only see the devices from Phoscon?

Inside deconz i have many devices, but i cannot see them.

is there any way to "un-filter" and get all devices from the app deconz and into domoticz?

Smanar, 2020-01-21:
Hello, 
You can see your device in Phoscon (the web app) but not in deconz (the desktop application) ?

For domoticz, if it works on phoscon/deconz, better to ask in the plugin section on domoticz forum. But for information, if it works on deconz, and if you use correct setting on the plugin, all the deconz devices will be visible on domoticz, there is no filter.

eddiex666, 2020-01-21:
Hi.

I can ses the devices in deconz app 

And not in phoscon web app :-)

I guess the phoscon app not support all devices.

Domoticz deconz plugin only ser the valid devices in phoscon.

Can this be fixed?

Smanar, 2020-01-21:
Ha ok so the devices are visible in deconz and domoticz but not Phoscon ?
If it's that, I m sorry, but it's something "normal", you are right Phoscon not support all devices and only deconz dev can do something.

closed 2020-05-29 by stale[bot]