dresden-elektronik/deconz-rest-plugin #2351: How do I debug the dev version?

mdolnik, 2020-01-21:
Hi there,

I have gotten the dev version of the rest-plugin working in ubuntu by using the [instructions on the home page](https://github.com/dresden-elektronik/deconz-rest-plugin#install-deconz-development-package-optional-linux-only) and it seems to compile properly as far as I can tell.

My question is what do I do next?

ebaauw, 2020-01-21:
Did you remove the stock rest api plugin from the `plugins` directory?  If you only rename it, it will still be loaded.

mdolnik, 2020-01-21:
After compiling I ran the `sudo cp ../libde_rest_plugin.so /usr/share/deCONZ/plugins` command which I've confirmed overwrites the `libde_rest_plugin.so` file in the plugins directory.

Is that what you mean by removing the plugin? Or is there another plugin file I need to remove?

ebaauw, 2020-01-21:
No, using `sudo cp` is good.

mdolnik, 2020-01-22:
I am brand new to the C++ and compiling world, so I can't say I know too much of what's going on, but I managed to get it working.

I tried running `make clean` then I ran `qmake && make -j2` and when I re-copied the modified `libde_rest_plugin.so` to `/usr/share/deCONZ/plugins` and ran `deCONZ --dbg-info=2` I am now getting the updated logging messages that confirm that my modified code is being used.

So `make clean` is what solved my issue, and I only needed to run it once. I'm not entirely sure why it worked, as looking into that command people seem to think it's not necessary, but it worked for me.

closed 2020-01-22 by mdolnik

ebaauw, 2020-01-22:
I run `make clean` every time after I’ve synced my local fork with the master.  Not sure if it’s needed, but it seems the prudent thing to do.

mdolnik, 2020-01-22:
Good to know, again I'm new to this so I'm not sure if it's common knowledge or not. Thanks for your help.