dresden-elektronik/deconz-rest-plugin #2352: Get Backup over bash via api-"curl"-call / post

ITSSeu, 2020-01-21:
I am using deCONZ through apt-get install for a few days and now beginn to play around with the api.

deCONZ is running on my RPI 4 with Buster and I want to be able, to run a light cron bash-script for my backup of deCONZ Phosocon APP and api. [This thread ](https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1307) give's me the only information about receiving the export via an api call.
 
I want to connect with curl to the api from the same machine as deCONZ is locally running on in bashline. For that, I already generate an apikey:
```
**command**: 
curl -k -S -v -H "Content-Type: application/json" --data @apikey_body.json http://localhost:9090 -X "POST /api"

**apikey_body.json**:
{ "devicetype": "NEWUSERNAME" }

**respone**:
[{"success":{"username":"B422CE84A1"}}]
```

When I call with this apikey to  /api/username/config/export I am receiving 4 files to ~/.local/share/dresden-elektronik/deCONZ:

```
**command**: 
curl -k -S -v -H "Content-Type: application/json" http://192.168.188.203:9090/ -X "POST /api/"NEWUSERNAME"/config/export"

**response**:
[{"success":{"/config/export":"success"}}]

ls ~/.local/share/dresden-elektronik/deCONZ
config.ini  session.default  zcldb.txt  zll.db
```

I think that the files are okay, but I find no way to import them via api. Only phoscom app get the option to restore a backup file in the .dat filetype. Like the one I can manually download from the phoscom app.

I already compress the files to tar file, also rename the tar file in .dat but the file can not import by phoscom app.

Can sombody help me please. I am very new in deconz, api and curl. Thanks in advanced :)

Jeroensum, 2020-02-24:
Have you tried renaming it to something similar to:
raspbee_gateway_config_2020-02-24.dat 

Also if I create an export by hand I do not get a config.ini file, do you?

Third a question from my side, I followed your instructions and I'm able to trigger the api to generate an export, it reports success, however I never receive the files, any idea? The api responds proper when I ask for things like lights so I know the username is good.

ebaauw, 2020-02-24:
> I think that the files are okay

Yes, these are the regular file containing the configuration; but no, this is **not** the backup.  The export should create `deCONZ.tar.gz` in the same directory.

```
$ ls -al .local/share/dresden-elektronik/deCONZ/
total 144
drwxr-xr-x 2 pi pi   4096 Feb 24 19:48 ./
drwxr-xr-x 3 pi pi   4096 Nov 22 12:15 ../
-rw-r--r-- 1 pi pi   2070 Feb 22 23:16 config.ini
-rw-r--r-- 1 pi pi   2125 Feb 22 23:15 session.default
-rw-r--r-- 1 pi pi     35 Nov 22 12:15 zcldb.txt
-rw-r--r-- 1 pi pi 122880 Feb 24 19:48 zll.db
$ ph -H 127.0.0.1:8008 post /config/export
"success"
$ ls -al .local/share/dresden-elektronik/deCONZ/
total 164
drwxr-xr-x 2 pi pi   4096 Feb 24 20:43 ./
drwxr-xr-x 3 pi pi   4096 Nov 22 12:15 ../
-rw-r--r-- 1 pi pi    614 Feb 24 20:43 AccessoryInfo.CC223DE3C930.json
-rw-r--r-- 1 pi pi    515 Feb 24 20:43 AccessoryInfo.CC223DE3CE30.json
-rw-r--r-- 1 pi pi   2070 Feb 22 23:16 config.ini
-rw-r--r-- 1 pi pi   8917 Feb 24 20:43 deCONZ.tar.gz
-rw-r--r-- 1 pi pi   2125 Feb 22 23:15 session.default
-rw-r--r-- 1 pi pi     35 Nov 22 12:15 zcldb.txt
-rw-r--r-- 1 pi pi 122880 Feb 24 19:48 zll.db
$ tar tvzf .local/share/dresden-elektronik/deCONZ/deCONZ.tar.gz 
-rw-r--r-- pi/pi           685 2020-02-24 20:43 deCONZ.conf
-rw-r--r-- pi/pi        122880 2020-02-24 19:48 zll.db
-rw-r--r-- pi/pi          2125 2020-02-22 23:15 session.default
-rw-r--r-- pi/pi           614 2020-02-24 20:43 AccessoryInfo.CC223DE3C930.json
-rw-r--r-- pi/pi           515 2020-02-24 20:43 AccessoryInfo.CC223DE3CE30.json
$ ls -al .homebridge/persist/
total 40
drwxr-xr-x 2 pi pi  4096 Nov 22 12:26 .
drwxr-xr-x 4 pi pi  4096 Feb 20 17:34 ..
-rw-r--r-- 1 pi pi   614 Feb 22 23:16 AccessoryInfo.CC223DE3C930.json
-rw-r--r-- 1 pi pi   515 Nov 22 12:26 AccessoryInfo.CC223DE3CE30.json
-rw-r--r-- 1 pi pi 19087 Feb 22 20:47 IdentifierCache.CC223DE3C930.json
-rw-r--r-- 1 pi pi  1176 Nov 22 12:26 IdentifierCache.CC223DE3CE30.json
```

@manup, @ChrisHae:
- Why is deCONZ trying to back up my homebridge configuration, when it doesn't manage homebridge?!
- Note that the homebridge backup is incomplete.  `~/.homebridge/persist` might contain multiple `AccessoryInfo` and `IdentifierCache` files, when homebridge exposes non-bridged accessories next to the bridge.  This is quite common: e.g. plugins exposing _Camera_ or _Television_ accessories, which cannot be bridged.

Jeroensum, 2020-02-24:
@ebaauw In your screendump I see you use ph, which I assume an alias of sorts? Can you please share your full command to trigger the export?

ebaauw, 2020-02-24:
`ph` is a little JavaScript program, distributed with homebridge-hue, to issue Hue or deCONZ API commands.  I always get lost in `curl` command line arguments ;-)

```
$ ph -h
Command line interface to Philips Hue or deCONZ API.

Usage: ph [-hVp] [-H hostname[:port]] [-u username] [-t timeout] command [argument ...]

Parameters:
  -h, --help
  Print this help and exit.

  -V, --version
  Print version and exit.

  -p, --phoscon
  Imitate the Phoscon app.  Only works for deCONZ.

  -H hostname[:port], --host=hostname[:port]
  Connect to hostname:80 or hostname:port instead of the default localhost:80.

  -u username, --username=username
  Use username instead of the username saved in ~/.ph.

  -t timeout, --timeout=timeout
  Set timeout to timeout seconds instead of default 5.

Commands:
  get [-hsnjuatlkv] [path]
  Retrieve path from bridge/gateway.

  put [-hv] resource [body]
  Update resource on bridge/gateway with body.

  post [-hv] resource [body]
  Create resource on bridge/gateway with body.

  delete [-hv] resource [body]
  Delete resource from bridge/gateway with body.

  discover [-hv] [-t timeout]
  Discover bridges/gateways.

  config [-hs]
  Retrieve bridge/gateway configuration (unauthenticated).

  description [-hs]
  Retrieve bridge/gateway description.

  createuser [-h]
  Create bridge/gateway API username.

  unlock [-h]
  Unlock bridge/gateway so new API username can be created.

  touchlink [-h]
  Initiate a touchlink.

  search [-h]
  Initiate a seach for new devices.

  lightlist [-hv]
  Create/update lightlist resourcelink.

  outlet [-hv]
  Create/update outlet resourcelink.

  probe [-hv] [-t timeout] light
  Probe light for supported colour (temperature) range.

  restart [-hv]
  Restart Hue bridge or deCONZ gateway.

For more help, issue: ph command -h

See https://github.com/ebaauw/homebridge-hue for more info.
(homebridge-hue v0.11.55, node v12.16.1)
```

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix