dresden-elektronik/deconz-rest-plugin #2353: gateway not yet initialised

mink2k, 2020-01-21:
I‘m very frustrated! I have Setup my pi a couple of times and every time I want to start homebridge I get this: 

`Jan 21 20:23:38 phoscon homebridge[3718]: [2020-1-21 20:23:38] [Hue] Phoscon-GW8: gateway not yet initialised - wait 1 minute
Jan 21 20:24:38 phoscon homebridge[3718]: [2020-1-21 20:24:38] [Hue] Phoscon-GW8: /resourcelinks/1: /lights/4: not available
Jan 21 20:24:38 phoscon homebridge[3718]: [2020-1-21 20:24:38] [Hue] Phoscon-GW8: gateway not yet initialised - wait 1 minute `

I have search google all the way but can’t find a answer that helps - all I do is crashing my system again and again but nothing repairs this issue. 
Can somebody help?

ebaauw, 2020-01-21:
Did google reveal this issue: https://github.com/ebaauw/homebridge-hue/issues/350?

mink2k, 2020-01-22:
Your my man! Thanks a lot! But no it dose not reveal this. 

Thank you thank you thank you!

closed 2020-05-29 by stale[bot]