dresden-elektronik/deconz-rest-plugin #2356: Xiaomi Honeywell smoke detector

Thomas1305, 2020-01-22:
Hi there,

will the Xiaomi honeywell smoke detector be supported in future? [https://de.gearbest.com/alarm-systems/pp_615081.html](url)
If i pair it in the phoscon app, it´s not shown in the device list. 
But in iobroker at the objects-tab it´s shown.
Why?
Please integrate it in the phoscon app too.
It would be greate.

SwoopX, 2020-01-26:
If you got it in iobroker, then it should already been supported. To swiftly verify, please provide the screenshots mentioned in the wiki for requesting device support and ensure, the attributes have been read.

peer69, 2020-01-26:
It’s supported via REST API, not via Phoscon. You can’t see it in the webinterface but it works using the GUI which is why iobroker shows it as would any other application using the api.

RezzZ, 2020-03-15:
problem is you can't easily rename your smoke detectors like you can with lights/sensors/switches.
would love to see this device added in deconz as well.
makes it easier to know which device is which in e.g. homeassistant instead of names like
lumi.sensor_smoke

jakubkinst, 2020-05-12:
I have the same problem. I am able to pair the device with Phoscon, but it's never added to the list of the sensors. The sensor is celarly added though - I can see it in Home Assistant and in the mesh GUI.

SwoopX, 2020-05-12:
@jakubkinst This is no problem then. DE has the sensor/device just not added to Phoscon then. It's closed source.

andriej, 2020-05-12:
And you cannot control it as you can in example in [z2m](https://www.zigbee2mqtt.io/devices/JTYJ-GD-01LM_BW.html) - self-check or sensitivity setting is not available in REST-API.

jihonrado, 2020-05-13:
What are you doing exactly to pair the smoke detector? I have the same model and I don't even see any logs in `deconz` when I press the button in the sensor as I do with other sensors. Thanks in advance.

jakubkinst, 2020-05-13:
@jihonrado just press the button on the sensor 3 times shortly

RezzZ, 2020-05-13:
the absense of additional control (selfcheck, sensitivity) is already discussed here:
https://github.com/dresden-elektronik/deconz-rest-plugin/issues/152#issue-254099237

I don't believe this issue is valid as it is a problem of phoscon which is independent from deconz-rest-plugin (I made the same mistake, and it's easy to believe they belong together when using e.g. the docker container)

jihonrado, 2020-05-13:
Thanks @jakubkinst. That worked! As mentioned, it does not show in the Phocon Web UI but it does in the deconz VNC UI.

Ageo, 2020-05-25:
Suggestion for developers: Could you add the Honeywell smoke detector as a light for example? Like the Ikea USB extender. For the moment, I'm not able to delete the smoke detector thought the App.

RezzZ, 2020-05-25:
> Suggestion for developers: Could you add the Honeywell smoke detector as a light for example? Like the Ikea USB extender. For the moment, I'm not able to delete the smoke detector thought the App.

The 'app' is phoscon. Closed source and different developer. You're request is not valid for the deconz rest api

SwoopX, 2020-05-25:
> You're request is not valid for the deconz rest api

That's not entirely correct. The REST API "determines" what your device will show up in the end (light, sensor, or both). It's then up to the developers to also do the integration for Phoscon. Now, the smoke detector is neither a light, nor does it have an IAS WD cluster and therefore, it is not exposed as light.
As said, it up to the developers to add it to Phoscon and for that, this is the wrong place indeed. However, device deletion has some room for improvement.

You can make a device leave (but with rejoin, not reset) in deconz GUI, check out https://github.com/dresden-elektronik/deconz-rest-plugin/wiki/deCONZ-for-Dummies
Eventually, deleting the light/sensor via REST API does reset it.

RezzZ, 2020-05-25:
you are right @SwoopX, but I can't believe the developers will spend time in developing a workaround to get smokedetectors visible in phoscon as a light.

@Ageo : support@dresden-elektronik.de. Perhaps if enough people ask for support they will grant our request.