dresden-elektronik/deconz-rest-plugin #2359: link button not pressed

tuxick, 2020-01-22:
When trying to connect with home-assistant i get:
pydeconz.errors.pydeconzException: / link button not pressed

It it totally unclear what this "link button" is. There's "Authenticate app", but that's not it, since i did try that one.

ebaauw, 2020-01-22:
It probably is that one.  On the old web app it's _Unlock Gateway_ on the _Settings_ page.  Make sure to press that one first, and start HA within 60 secs.

tuxick, 2020-01-22:
OK, the problem was HA had settings from previous setup in .storage/core.config_entries
After removing the deconz section all worked well. So the error message is rather misleading :)

closed 2020-05-30 by stale[bot]

poelzi, 2022-02-13:
I got the same issue when upgrading my deconz docker container to the new repo and forgot to change the data volume.
The integration shows a reauthenticate button but reauthentication does not work.