dresden-elektronik/deconz-rest-plugin #2360: New device support request: Tuya/Samotech/... motion sensor

jdlondon, 2020-01-22:
Requesting support for the various motion sensor devices that re-badge Tuya's standard offering (such as Samotech SM301Z)

Debug shows the following messages during enrol and motion status change:
```
19:40:45:799 IAS Zone Enroll Request, zone type: 0x000D, manufacturer: 0x1002
19:40:45:953 IAS Zone Status Change, status: 0x0000, zoneId: 100, delay: 0
19:40:47:164 IAS Zone Status Change, status: 0x0000, zoneId: 100, delay: 0
19:40:56:564 IAS Zone Status Change, status: 0x0001, zoneId: 100, delay: 0
19:40:57:777 IAS Zone Status Change, status: 0x0001, zoneId: 100, delay: 0
19:41:57:156 IAS Zone Status Change, status: 0x0000, zoneId: 100, delay: 0
19:41:58:367 IAS Zone Status Change, status: 0x0000, zoneId: 100, delay: 0
19:42:15:611 IAS Zone Status Change, status: 0x0001, zoneId: 100, delay: 0
19:42:16:822 IAS Zone Status Change, status: 0x0001, zoneId: 100, delay: 0
```

[image]
[image]

Many thanks!

closed 2020-05-30 by stale[bot]

BergenLarsen, 2020-06-05:
How can I return this to an open issue and help you guys work on this?

hastime, 2020-06-21:
Can this be re-opened?

SwoopX, 2020-06-21:
There's no reason. Device is supported and Phoscon is not the REST API plugin.

You may raise an issue at the Phoscon beta repository to have them visible.

hastime, 2020-06-21:
Hi, 

Thanks for the quick response! I will raise it at the Phoscon beta repo! I am amazed at how well Deconz works, keep up the good work!

Callum17, 2020-07-20:
Currently trying to pair the same sensor via Deconz and Home Assistant. Deconz is failing to return any cluster information for me. Running 2.05.79 via docker armhf: https://hub.docker.com/r/marthoc/deconz.

@jdlondon / @hastime - What release and platform were we using?