dresden-elektronik/deconz-rest-plugin #2361: combee II key disconecting

jpmimard, 2020-01-22:
I've the key since 5 days.
It works good on RPI 3 days, and after on freebox Delta (jeedom) 1 Day, but now it's seems to be broken.
Linux report connection/disconnection of the key again and again
I try on windows, same, song for key insertion, song for key removing, again and again...
4 days for a key...before death

Smanar, 2020-01-22:
I have see some similar problem on jeedom forum, your key is direcly on the freebox ? can you make a try with a powered USB HUB (if possible usb 2.0) ?

jpmimard, 2020-01-23:
I'd tried:
direct on the box,
   on USB A direct on the box
   with USB A Cable extension
   on USB C  with Hub USB 3 powered and non powered
on laptop USB A  2 and 3

On the box linux see the key, SN and type and after 1 or 2 second loose the connection and start again.
I think the key power circuit is dead

I will buy an other model

Smanar, 2020-01-23:
Contact the support first at support@dresden-elektronik.de they will ask for some tests but just after some days I think they will propose an exchange.

closed 2020-05-30 by stale[bot]