dresden-elektronik/deconz-rest-plugin #2362: SD to SSD went bad for my Conbee 2

Xavoo, 2020-01-23:
’m a conbee 2 user, and all worked fine for me until yesterday. 

I got a raspberry pi 3, and Jeedom

I installed Jeedom on a SD card, and two weeks ago I moved everything on a ssd drive, for more stability. 

in order to get everything clean, I just took a jeedom backup, and I imagined that my conbee would have followed. But not …

When i go to phoscon app, i just see 2 gateways, obviously there are the same … 

I can’t reset any of the gateways, i just got a failure every time. Same, I tried to make a backup, failure every time also …

How can I do ? I just have a conbee that is completely unusable today, and no way to reset it, from ssd system or SD. 

Since this day, I tried a lot of things : reset raspberry, reset jeedom, tried to connect conbee with my former SD card, Nothing works. On phoscon app, the details about the gateway are almost all blank. Vendor : unknown, Product (Blank), Version : 2.05.71 17/11/2019, Firmware : not connected. 

Jeedom restoration just helped me not to have an error message about ip address. Now it seems just to be unable to detect new sensors or lights ... ID into the app is 0000000000000000. 

I'm just lost ! 

Thank you for your reply !

Smanar, 2020-01-23:
First thing, you need to have a firmware version displayed.

Have you tried to make a clean deconz re-installation (using jeedom not command line)?
Here you have some tips for the "firwmare not connected"
https://phoscon.de/en/support#conbee2-connection-issues

ebaauw, 2020-01-23:
Be sure to connect the SSD to a USB2 port.  USB3 causes radio interference in the 2.4 GHz band that ZigBee uses.

Xavoo, 2020-01-23:
Wow, quite strange. I tried multiple times the whole procedure Smanar sent, nothing worked. Even $ lsusb was disappointing because my conbee wasn't even detected. The update was tried many times. 

I was writing this comment, and, guess what, everything went good, with no major change into my procedure ... 

Anyway thanks for your help :)

closed 2020-01-23 by Xavoo