dresden-elektronik/deconz-rest-plugin #2365: Is it possible to configure a ConBee2/deCONZ ZigBee Network whithout the phoscon webinterface?

Andreyooo, 2020-01-23:
The REST API only supports lighting so far. Is it possible to configure the ZigBee Network without a graphical interface? Is there an API for manual communication?

Smanar, 2020-01-23:
Depend of feature you want, but I think all phoscon feature can be used with API.
Open network, change channel, ect ...

Smanar, 2020-01-23:
Depend of feature you want ?
But I think all you can make with phoscon can be made with API, open network, change channel, ect ...

ebaauw, 2020-01-23:
Phoscon is just another API client; deCONZ runs perfectly well without it.  The link to the API documentation is in the README.

Andreyooo, 2020-01-23:
Oh I see, the REST API does support all kinds of devices, I misread the repo title.
So this means I can access all kind of ZigBee 3.0 devices through the API, even devices that the Phoscon webinterface does not support yet?

ebaauw, 2020-01-23:
Yes.  The REST API plugin is open source, so we can add API support for new devices ourselves.  Phoscon isn’t and can only be updated by dresden elektronik, so adding support for new devices typically takes a bit longer.

Andreyooo, 2020-01-23:
Okay, ty very much :)

closed 2020-01-23 by Andreyooo