dresden-elektronik/deconz-rest-plugin #2367: LEDVANCE 5/6 Inch Retrofit RGBW and FLEX RGBW - Incorrect type in Hue bridge

kq9914, 2020-01-23:
This is similar to https://github.com/dresden-elektronik/deconz-rest-plugin/issues/1835 but for different light models and full color bulbs vs color temperature. 

I have the Lightify/LEDVANCE 5/6 inch retrofit RGBW lights (RT RGBW) and the RGBW strip (FLEX RGBW) that show up in the Hue bridge as "Color dimmable light" rather than "Extended Color Light", which results in OpenHAB not detecting them via the Hue bridge interface.

Can this be updated?  Let me know if more information is needed.  API output for one of each is below.

`    "14": {
        "ctmax": 666,
        "ctmin": 153,
        "etag": "50e995d7a0e3cacfad12dbe19be8de90",
        "hascolor": true,
        "manufacturername": "LEDVANCE",
        "modelid": "FLEX RGBW",
        "name": "Bathroom Under Counter",
        "state": {
            "alert": "none",
            "bri": 0,
            "colormode": "ct",
            "ct": 271,
            "effect": "none",
            "hue": 0,
            "on": false,
            "reachable": true,
            "sat": 254,
            "xy": [
                0.5064,
                0.1674
            ]
        },
        "swversion": "00102101",
        "type": "Color dimmable light",
        "uniqueid": "f0:d1:b8:00:00:07:25:a2-01"
    },
    "2": {
        "ctmax": 526,
        "ctmin": 153,
        "etag": "2576d190715275eef5f1a8c1f9c8ca8b",
        "hascolor": true,
        "manufacturername": "LEDVANCE",
        "modelid": "RT RGBW",
        "name": "Living Front Left",
        "state": {
            "alert": "none",
            "bri": 77,
            "colormode": "ct",
            "ct": 341,
            "effect": "none",
            "hue": 0,
            "on": true,
            "reachable": true,
            "sat": 254,
            "xy": [
                0.735,
                0.265
            ]
        },
        "swversion": "00102101",
        "type": "Color dimmable light",
        "uniqueid": "f0:d1:b8:00:00:06:b1:d6-01"
    },````

Thanks,
James

kq9914, 2020-04-23:
Hello!  Wanted to follow up on this.  Let me know if any additional information is needed.  Thanks!

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix