dresden-elektronik/deconz-rest-plugin #2368: FYRTUR IKEA after restart not reachable

Michaelnorge, 2020-01-23:
The window shutters from IKEA are unavailable after restart.
I am using the latest version of Deconz.
That is not shown in the Phoscon-App (there they are still online, but not assessable.

But in ioBroker the indicator show "FALSE"

Maybe time to give the shutters new icons as well, they coming up with a lamp-symbol
[image]

handy80, 2020-01-24:
Hi Michaelnorge, I cannot help you but can you point me to a guide how I can get my fyrtur set (blind, switch and repeater) connected to deconz/phoscon I would like to control the blind with the switch and by ioBroker as you but I unfortunately don't know how. The set components are paired all right and work well.

Michaelnorge, 2020-01-24:
Hello.
You don¨t need that stuff, just the shutters.
Press both buttons on the shutters about 5 seconds until the light starts turning on.
Let ioBroker find "lamps" - finish!

greetoz, 2020-01-25:
I have a similar problem. After a restart from deconz the blinds are not always reacting. I have to go into the VNC GUI, Go into basic cluster info and press the READ attributes then the manufacturer strings etc get filled in and the blind is usable again.

Before pressing the read button
[image]

After pressing the read button
[image]

Michaelnorge, 2020-01-25:
I could¨t get installed VNC GUI for some reason.
I just reconnect the shutters by pressing both buttons of the shutters.

But it is strange that I have three shutters, but just two of them are disconnected after reboot.

handy80, 2020-01-26:
thanks, tried that and stuck in the same situation (even if I messed up and paired twice the old light shows reachable = 'true' and the new shows 'false'
Both lights are not working from within ioBroker and the PhosconApp.
Any news here?

ginkel, 2020-01-27:
Are the blinds connected directly to the ConBee or to another router?

Michaelnorge, 2020-01-27:
> 
> 
> Are the blinds connected directly to the ConBee or to another router?

Directly...

Smus, 2020-03-02:
I have the exact same issue. I connect Ikea fyrtur blinds to conbee2, everything work.. restarts and all is lost again

Michaelnorge, 2020-03-03:
... but its easy to get them connected again. Just click on "pair" in the Phoscon App and go around and keep pushing the two buttons on the shutters until they start moving. 
Its even possible to pair a lot of them at the same time :-)

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix