dresden-elektronik/deconz-rest-plugin #2369: Subclass some request from a device to the coordinator.

Smanar, 2020-01-23:
Hello, I have a device that make request to the coordinator.
- Read attribute cluster 0000 attribute F0000
- Set reporting with timeout

And the conbee doesn't answer, even with error message like other gateway.

There is something in the API I can edit to make my own answer to theses request or it's only possible editing the deconz code ?

ebaauw, 2020-01-23:
> Read attribute cluster 0000 attribute F0000

That's a hex digit too many in the attribute id.  What endpoint on the gateway does it send the request to?

> Set reporting with timeout

What command does it send?  Do you have a sniffer capture?

> And the conbee doesn't answer, even with error message like other gateway.

I wouldn't expect the ConBee firmware nor the deCONZ core programme to answer.  The REST API plugin can be extended to do so, though.  We've implemented the server _Time_ cluster for the coordinator, because some Xiaomi devices expect the coordinator to serve that.

Basically, you'd need to do the following:
- Configure the core programme to advertise the _Basic_ cluster in the _Simple Descriptors_. In the GUI, open the _Network Settings_ window, select the _Endpoints_ tab, and add 0x0000 to the _In Clusters_.  Double-check that the drop-down menu now shows the (blue) _Basic_ cluster.  Note that you cannot query any of the attributes from the _Cluster Info_ panel yet.
- Define a `handleBasicClusterIndication()` in `de_web_plugin_private.h`:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/10810836905fbcd1e7c9fafcaf386702c9fed41f/de_web_plugin_private.h#L1309
- Add a `case` to handle an APS indication for the _Basic_ cluster to `apsdeDataIndication()`:
https://github.com/dresden-elektronik/deconz-rest-plugin/blob/10810836905fbcd1e7c9fafcaf386702c9fed41f/de_web_plugin.cpp#L574
- Add `basic.cpp` to `SOURCES` in `de_web.pro`;
- Copy `time.cpp` to `basic.cpp` and adjust for the _Basic_ cluster.  It might be enough to only implement the `default` case and return an _Unsupported Attribute_ for any attribute, but you could knock yourself out and implement _Manufacturer Name_, _Model Identifier_, _SW Version_ (with the device firmware version), and maybe some more.  Test with the GUI, in the _Cluster Info_ panel.

Smanar, 2020-01-23:
Yeah, I have just tested the change, tomorrow I will make the code changement, thx a lot.

For screenshoots
[image]
This one is the more important, theses devices can pair as it without pressing a reset button so the manufacture have make a security, to pair a device you need to power off/on all the network. this value is reset after the power cycle, so when a device join the network it ask for this attribute, if the value is too high , it leave the network. This procedure is from zigbee2mqtt project.

I can't explain why, but without it, it hard for pairing, but It works. Zigate use it too. And as I have some problem ATM with some devices I m trying all possibilities.

[image]

This one is less important (I think) because other gateway answer "unsupported general command 0x82". And I m seeing how to make with your code for the read attribute, but to answer to a "set attribute report" ....

Smanar, 2020-01-24:
Yeah it's working now.

Without this modification, the pairing is possible, but the device works bad.
I have it full working again now.

Thx a lot @ebaauw ^^, 3 weeks I m on this bug. A big mystery.

Now need to find if it's possible adding this cluster in deconz config, lot of user use headless version, so it will hard for them adding this option.

Smanar, 2020-01-24:
BTW, if I don't change the deconz configuration, It don't work with GUI but still work with API ?

ebaauw, 2020-01-24:
> Now need to find if it's possible adding this cluster in deconz config, lot of user use headless version, so it will hard for them adding this option.

I'd tell them to enable the GUI, but if you like to live dangerously...

The setting is stored in the `device_descriptors` table in the SQLite database, in the row with `device_id` 1 and `endpoint` 1.  I think the `device_id` will always be 1, as the RaspBee or ConBee has to be the first node created by deCONZ, but maybe some backup/restore scenarios might cause an exception.  You might want to lookup the `device_id` from the `devices` table, based on the RaspBee/ConBee mac address.

The `data` BLOB seems holds the _Simple Descriptor_ for the endpoint.  It seems quite straightforward to decode (note the big endianness):
```
01 04 01 05 00 01 02 0a 00 19 00 01 00 05
 |     |     |  |  |     |     |  |     + out (client) cluster: 0x0500 (IAS Zone)
 |     |     |  |  |     |     |  +------ # out (client) clusters: 1
 |     |     |  |  |     |     +--------- in (server) cluster: 0x0019 (OTAU)
 |     |     |  |  |     +--------------- in (server) cluster: 0x000a (Time)
 |     |     |  |  +--------------------- # in (server) clusters: 2
 |     |     |  +------------------------ device version: 1 
 |     |     +--------------------------- device type: 0x0005 (Configuration Tool)
 |     +--------------------------------- profile: 0x0104 (ZHA)
 +--------------------------------------- endpoint: 0x01
```
I haven't tried modifying the BLOB myself, but I wouldn't be surprised if it would work.  Of course the usual caveats apply: make sure to quit deCONZ and to backup the database, before touching it.

ebaauw, 2020-01-24:
> BTW, if I don't change the deconz configuration, It don't work with GUI but still work with API ?

I don't think so.  I think the deCONZ core only forwards APS messages to the REST API plugin for clusters that are in the node descriptor.  I think that's why we had to push back the _IAS Zone_ cluster to certain sensor that didn't advertise it, see also https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2061#issuecomment-577319666.  Clarke's Third Law definitely applies here.

Smanar, 2020-01-24:
Lol, nope I don't relay like danger as I working on my production network, I have been hearing since 3 week, all days :  "why the dimmer is still broken, and nianiania". So now I will stop all my tests, and I don't think manu will be agree with that too.

But fortunately, it works ^^, I have just make tries, you haven't the cluster on the GUI, but the code is still working, so perfect news for me.

ebaauw, 2020-01-24:
That's good news indeed.

closed 2020-01-29 by Smanar