dresden-elektronik/deconz-rest-plugin #2370: Tuya zigbee temperature sensor and Tuya zigbee motion sensor

jahodaradek, 2020-01-23:
Tuya motion
[image] 
Tuya motion
[image]
Tuya temperature
[image] 
Tuya temperature
[image] 

Sorry for the czech language but the you can see they are like xiaomi sensors just with tuya platform https://www.chytrevypinace.cz/ZigBee-PIR-Sensor-Pohybu-d155.htm
https://www.chytrevypinace.cz/ZigBee-Sensor-Teploty-a-Vlhkosti-d154.htm
Also they work with ZHA.

gesamtplan, 2020-01-23:
I just added the temperature sensor to my rasbee and get data from it like described above. I am not so firm with the deconz app but I don't see the sensor in the phoscon app. Is this normal or am I making something wrong? 

Some lamps and the Eurotronic thermostats are working fine with my Rapsberry Pi 3.

jahodaradek, 2020-01-23:
Yup can be seen in deconz but not phoscon.

gesamtplan, 2020-01-23:
Is there any chance to get them integrated in home assistant?

SwoopX, 2020-01-24:
Folks, the devices will not be visible in Phoscon unless DE specifically integrates them. To see if they are available, check the REST API.

I did integrate one of those some weeks ago under a different name.

gesamtplan, 2020-01-24:
How can I do this? Maybe i can get the adapt the rest api on HomeAssitant.

SwoopX, 2020-01-25:
You may check it following this here https://github.com/dresden-elektronik/deconz-rest-plugin/issues/2121#issuecomment-564770517

What exactly do you want to adapt? Either the device ist supported and the sensors are being created or not. The API has pretty much nothing to do with it.

jahodaradek, 2020-01-26:
I guess he wanted to ask if its possible to maybe create sensors by himself before the device is officially supported.
I tried to compile your branch and replace the plugin but got into some errors 
[image]

SwoopX, 2020-01-26:
Bug is fixed

gesamtplan, 2020-01-27:
I am not so firm in compiling c-projects. I just saw them on the deconz gui and not in the phoscon app. So if the device is not supported yet it explains why there ist no sensor in the rest-api. Got that.

SwoopX, 2020-01-27:
> I am not so firm in compiling c-projects. I just saw them on the deconz gui and not in the phoscon app. So if the device is not supported yet it explains why there ist no sensor in the rest-api. Got that.

I'd rather put it that way:
If a device is **not** supported, you should see it in deconz GUI, but must likely sensors are not created in REST API.
If a device **is** supported, you see it in deconz GUI, sensors should be available through REST API but the device is not necessarily showing up in Phoscon.

gesamtplan, 2020-01-27:
Is there a chance that the sensors are getting supported in a newer version of deconz?

SwoopX, 2020-01-27:
I've already issued a pull request.

gesamtplan, 2020-02-04:
Works good in my environment. Three days for now without failure

SwoopX, 2020-02-04:
Sounds good to me. Do you get proper values for battery? I'd assume not and I probably need to amend it for other devices as well.

gesamtplan, 2020-02-04:
Have checked the battery levels of my two sensors. One shows about 52% the other 74%. I don't know how long the sensors have been in stock in china and the batteries haven't been sealed as I got them. 

Have just added a new battery in in on of the sensors and it still shows 52% How did u calculate?

gesamtplan, 2020-02-04:
Temperature and Humidity are looking quite good and do not differ as much from a other sensor I use. Temperature compared 18.2 degC to 17.8 degC and humidity 51,4% to 47.1%. I don't believe what's behind the dot. Measuring temperature and humidity is quite complicated to get proper values. But it doesn't sound very implausible.

SwoopX, 2020-02-04:
Interesting. Battery level is not properly displayed for a couple of sensors I previously integrated, that's why I was asking. But good to see that it's working on your end.

If you can provide a screenshot from power configuration cluster with read attributes, I can tell you. Right now, I assume it is based on battery voltage, not too accurate. Depending on when you exchanged the battery, it might take some time for the value to refresh.

gesamtplan, 2020-02-04:
Had exchanged the battery and refreshed via the button on the sensor. HomeAssistant showed a refresh less then a minute. Maybe the sensor needs some time to middle the values or measure the new battery. I will look tomorrow. Thanks for ur help.

SwoopX, 2020-02-04:
Np. Could you share the screenshot anyway? Thanks!

gesamtplan, 2020-02-04:
I have to put a samba share on the Raspberry to do this. Maybe I have time tomorrow. But the battery level on the sensor increased to 100% a few minutes ago. :-)

SwoopX, 2020-02-04:
Ok, that's a bit too much then. Just lemme know the values of attribute 0x0020 and 0x0021. Thanks and glad that it worked out so smoothly!

gesamtplan, 2020-02-05:
The sensor with 100% shows on 0x0020 the value 30 and on 0x0021 the value 200. The second one shows on 0x0020 28 and on 0x0021 144. The RestAPI shows on the second one 72% which is half the hex value. Sounds plausible to me.

SwoopX, 2020-02-05:
Jep, all good. Got to the wrong line of code here so great it all works.

gesamtplan, 2020-02-05:
Where was the bug und have u checked in it yet?

abwyatt, 2020-02-17:
I would love to try the commit for the Tuya motion sensor. I've tried to compile from master using the instructions provided in the install instructions but I'm running into the following 2 errors with deconz.h. Any idea how I can address them? 

homeseer@Intel-Lubuntu-HomeSeer:~/deconz-rest-plugin$ qmake && make -j2
make -f Makefile.Release
make[1]: Entering directory '/home/homeseer/deconz-rest-plugin'
/usr/lib/x86_64-linux-gnu/qt5/bin/uic de_web_widget.ui -o ui_de_web_widget.h

... snipped this bit out

In file included from bindings.cpp:11:0:
de_web_plugin.h:16:20: fatal error: deconz.h: No such file or directory
compilation terminated.
Makefile.Release:488: recipe for target 'release/bindings.o' failed
make[1]: *** [release/bindings.o] Error 1
make[1]: *** Waiting for unfinished jobs....
In file included from authorisation.cpp:13:0:
de_web_plugin_private.h:24:20: fatal error: deconz.h: No such file or directory
compilation terminated.
Makefile.Release:470: recipe for target 'release/authorisation.o' failed
make[1]: *** [release/authorisation.o] Error 1
make[1]: Leaving directory '/home/homeseer/deconz-rest-plugin'
Makefile:38: recipe for target 'release' failed
make: *** [release] Error 2
homeseer@Intel-Lubuntu-HomeSeer:~/deconz-rest-plugin$

abwyatt, 2020-02-17:
Compiled from master and testing Tuya RH3040 Motion Sensor. Not appearing in Phoscon, but created sensor and battery device OK in JowiHue on HomeSeer.

SwoopX, 2020-02-18:
@abwyatt Thanks for your feedback. Version .73 has officially been released since a few days. The device will not show up in Phoscon unless specifically implemented. This must be done by DE as it is closed source.

CarpeDiemRo, 2020-05-10:
I have a PIR RH3040 and I have the following issue. 
It is successfully integrated into HA via DECONZ, not displayed in the DECONZ UI interface but appears as an entity in HA. So far, so good. 

I observed the following unnatural behavior.
a. The PIR doesn't detect motion continuously. EX. Detect the first movement and, even though the movement Is continuous in front of the sensor the PIR state turns to "off"
b. After this transition, the PIR doesn't change to 'on' even though the movement reappears in front. If I press the side button the sensor sent the message to DECONZ and it is triggering again, but only for a short period of time. 
c. It is clear that the relationship with DECONZ as a coordinator is ok and has no problem. My assumption is maybe the PIR is broken. Do you have the same issue? Or maybe I did something wrong. 
[image]

Mimiix, 2020-06-06:
It seems this issue is resolved or otherwise inactive. If it is not, please re-open!

closed 2020-06-06 by Mimiix

Zazou89, 2021-03-01:
Hello, 

I bought a Tuya smart Temperature sensor (zigbee version). 

Visible in Deconz but not in Phoscon. By extension not visible by Homebridge Hue. If I understand correctly there are no solution to display the sensor on Homebridge ? 

Thanks

Skare69, 2021-04-22:
I'm currently running version 2.10.04 / 14.3.2021 with firmware 26660700 and the Tuya temperature and humidity sensor is only visible in Deconz but not in the Phoscon App. It shows up in Home Assistant but as I can't change the name in Phoscon I have to change the device name in HA which is a bit tedious for a lot of sensor. 
I'd take a look at the code to fix this if anyone could give me a hint where to look (for the sensor not showing up in Phoscon).

Mimiix, 2021-04-22:
@Skare69 phoscon needs to add support for it themselves.

You can use the rest api to change the name.