dresden-elektronik/deconz-rest-plugin #2372: Deconz recognizes Sengled bulbs but can't control them

ultimateguy, 2020-01-24:
I have some Sengled E11-G13 bulbs that I am trying to add to Hassio using the Deconz addon and Conbee II stick. The bulbs are recognized and added in the deconz software, but afterward, trying to turn them on/off just makes the light blink for a few seconds.

All bulbs of this model seem to behave the same way. However I have other Sengled bulbs of different model and they work fine.

Any ideas?

Smanar, 2020-01-24:
Hello, do you have some logs ? even from HASS, just to see if it's not from a scenario/script/setting ?
For exemple websocket logs.

ultimateguy, 2020-01-24:
I can definitely get the logs after work tonight, but I'm not too familiar with logging. I gather from the docs that I can get logs by adding this in my config.yaml?

`logger:
  default: info
  logs:
    pydeconz: debug
    homeassistant.components.deconz: debug`

If there's another type of log you need please let me know. :)

ultimateguy, 2020-01-25:
So here are the logs starting when I try to add the Sengled light bulb. The bulb has id = 21.

```
2020-01-24 17:58:40 ERROR (MainThread) [pydeconz.websocket] Unexpected error 'light'
2020-01-24 17:58:40 DEBUG (MainThread) [pydeconz.websocket] Websocket starting
2020-01-24 17:58:40 DEBUG (MainThread) [pydeconz.websocket] Reconnecting to deCONZ in 15.
2020-01-24 17:58:55 DEBUG (MainThread) [pydeconz.websocket] Websocket starting
2020-01-24 17:58:55 DEBUG (MainThread) [pydeconz.websocket] Websocket running
2020-01-24 17:59:40 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"21","r":"lights","state":{"on":true,"reachable":true},"t":"event","uniqueid":"b0:ce:18:14:03:0f:7c:9a-01"}
2020-01-24 17:59:40 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"65520","r":"groups","state":{"all_on":false,"any_on":true},"t":"event"}
2020-01-24 17:59:40 DEBUG (MainThread) [pydeconz.websocket] {"e":"changed","id":"21","r":"lights","state":{"bri":255,"on":true,"reachable":true},"t":"event","uniqueid":"b0:ce:18:14:03:0f:7c:9a-01"}
```

ultimateguy, 2020-01-25:
I just discovered that clicking the "All Off" button in Phoscon turns off this Sengled light bulb properly. However whenever I try to control the bulb individually it just starts blinking instead of turning on or off properly.

ultimateguy, 2020-01-25:
And the plot thickens. If I add the bulb to its own group and turn that group on/off, the command works properly. Looks like I found a solution to a strange problem!

Smanar, 2020-01-25:
Yep there is nothing special on the log, except the line about the "all group "group > {"all_on":false,"any_on":true}

On my side alone bulb never trigger group event.

zagnuts, 2020-03-09:
FWIW I just got a couple of Sengled "Element Classic" bulbs. First odd behaviour was the first bulb was recognised as a dresden device with latest firmware, the second as 'unknown'. Both showed in the gui as being available - but turning them on/off or dimming did nothing. I would have tried to investigate further, but the concerning thing I found was both seemed to be getting quite hot around the base. Long term that would not be good for longevity, so I suspect I'll return them to the shop.....

Mimiix, 2020-06-05:
It seems this issue is inactive / resolved. If it is not, please re-open or create another issue!

closed 2020-06-05 by Mimiix